Mobsya/aseba #14: Complete asebahttp integration

MobsyaBot, 2018-04-17:
[image] **Issue by [davidjsherman](https://github.com/davidjsherman)**
_Saturday Aug 15, 2015 at 10:08 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/400_

----

Add menu entry for "Web Tools for Thymio" that runs asebahttp with the thymio_motion.aesl file parameter, the default TCP port 3000 (for the HTTP interface), and ser:name=Thymio-II; and asks user's web browser to open the javascript welcome page.

Depends on the following issues in https://github.com/davidjsherman/aseba/milestones/Version%201.4:
- davidjsherman/aseba#8
- davidjsherman/aseba#7
- davidjsherman/aseba#6

See also issue #401

MobsyaBot, 2018-04-17:
[image] **Comment by [davidjsherman](https://github.com/davidjsherman)**
_Sunday Aug 16, 2015 at 08:54 GMT_

----

After thinking about it, I believe that it is reasonable that both the web interface (davidjsherman/aseba#8) and Aseba Studio can communicate with the robot at the same time. If _asebahttp_ is run as described above, it will lock the USB connection and prevent Aseba Studio from communicating with the robot.

Consequently, I think that a better solution would be that the menu entry run _asebamedulla_ in addition to _asebahttp_ as described above. In that way Aseba Studio can share the connection to the robot using dashed target tcp:;port=33333, and _asebahttp_ will relay Aseba tcp:;port=33333 to HTTP localhost:3000.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Sunday Aug 16, 2015 at 19:59 GMT_

----

I agree but see two open questions:
- Should one run _asebamedulla_ or _asebaswitch_? To former provides a DBus bridge but is heavier and seems to have some bugs while the later is simpler. Also, _asebamedulla_ does not run on Windows.
- If we answer switch to the previous question, shouldn't the switch feature be provided by _asebahttp_ itself?

MobsyaBot, 2018-04-17:
[image] **Comment by [riedo](https://github.com/riedo)**
_Tuesday Aug 18, 2015 at 07:03 GMT_

----

asebamedulla is currently not packaged for Mac OS either. Why not use asebaswitch?

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Tuesday Jan 26, 2016 at 16:51 GMT_

----

Asebahttp is now working with protocol 5 (see 31d63747853e5ed9c5c3e215102730c887d7afec), so I retarget this issue for Aseba 1.6.

MobsyaBot, 2018-04-17:
[image] **Comment by [davidjsherman](https://github.com/davidjsherman)**
_Thursday Jan 11, 2018 at 15:53 GMT_

----

Asebahttp acts as an asebaswitch since c682c2103d6e10d4c65903f541703af2abfdbf55.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Feb 02, 2018 at 14:12 GMT_

----

Do we still want this in addition to the launcher (#728)?

MobsyaBot, 2018-04-17:
[image] **Comment by [cor3ntin](https://github.com/cor3ntin)**
_Friday Feb 02, 2018 at 14:25 GMT_

----

I'd say put all the issues vaguely related to the launcher or the switch in the launcher milestone and we will cleanup from there as we refine the use cases/ ux of the launcher ( and no, I don't think we want that, even though it's actually done on osx/win32, it will need to be improved, a lot )

cor3ntin, 2018-08-15:
This is no longer relevant as the thymio device manager replaces the http switch

closed 2018-08-15 by cor3ntin