Mobsya/aseba #15: Ubuntu packages have too large install footprint

MobsyaBot, 2018-04-17:
[image] **Issue by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Thursday Sep 17, 2015 at 10:54 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/426_

----

Currently, the PPA built of Aseba have an install footprint of more than 300MB, this is too much!

Things can be improved by:
- using shared libraries for Aseba
- not installing example binaries and experimental programs such as:
  - aseba-qt-dbus
  - aseba-qt-gui
  - aseba-test-natives-count
  - asebabotspeak
  - asebatest

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Monday Jan 25, 2016 at 14:52 GMT_

----

Switching to dynamic libraries is too much work for the short 1.5 deadline, pushing to 1.6

MobsyaBot, 2018-04-17:
[image] **Comment by [cor3ntin](https://github.com/cor3ntin)**
_Tuesday Feb 06, 2018 at 09:23 GMT_

----

the following libraries are good candidates for shared libraries

```
3,3M    aseba/clients/studio/libasebaqtcommon.a
3,1M    aseba/clients/studio/plugins/libasebaqtplugins.a
2,0M    aseba/common/libasebacommon.a
1,3M    aseba/common/zeroconf/libasebazeroconf.a
2,5M    aseba/compiler/libasebacompiler.a
```

MobsyaBot, 2018-04-17:
[image] **Comment by [davidjsherman](https://github.com/davidjsherman)**
_Tuesday Feb 06, 2018 at 14:46 GMT_

----

But not `vm` and `vmbuffer`, *cf.* https://github.com/aseba-community/aseba/issues/795#issue-291022563?

MobsyaBot, 2018-04-17:
[image] **Comment by [cor3ntin](https://github.com/cor3ntin)**
_Tuesday Feb 06, 2018 at 14:48 GMT_

----

@davidjsherman meh, they are extremely small libraries, a few K, no real need for them to not be static.

MobsyaBot, 2018-04-17:
[image] **Comment by [davidjsherman](https://github.com/davidjsherman)**
_Tuesday Feb 06, 2018 at 17:12 GMT_

----

Please document the use cases where more than one or two processes will be using the shared libraries that you have identified.

cor3ntin, 2019-03-15:
Closing that as we are moving towards flatpack

closed 2019-03-15 by cor3ntin