Mobsya/aseba #17: Scenario chooser in Playground

MobsyaBot, 2018-04-17:
[image] **Issue by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Wednesday May 11, 2016 at 09:30 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/506_

----

Playground should propose a nice scenario chooser at start, with for each scenario a cute screenshot and a description, translated in local languages. These scenarios should be packaged as resources in the application, to avoid installation path problems.

MobsyaBot, 2018-04-17:
[image] **Comment by [mbonani](https://github.com/mbonani)**
_Wednesday Jan 31, 2018 at 08:35 GMT_

----

as #728 development will provide something similar  i wonder if it is still usefull or it as to be used by the launcher

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Wednesday Jan 31, 2018 at 08:36 GMT_

----

We could still have a preview and data of a scenario when one is selecting a scenario, as Playground is not limited to Thymio. I want to keep it, but remove it from the milestone 1.7.

mbonani, 2019-04-26:
is this not a wrong closed even if scenario chooser is in launcher?

MobsyaBot, 2019-04-26:
Yep you can reopen