Mobsya/aseba #19: Playground scene format should be documented

MobsyaBot, 2018-04-17:
[image] **Issue by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Wednesday May 11, 2016 at 10:31 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/511_

----

Scene format should be documented, to allow third-parties to build scenes. This documentation should be translated.

MobsyaBot, 2018-04-17:
[image] **Comment by [ddoak](https://github.com/ddoak)**
_Wednesday May 11, 2016 at 16:54 GMT_

----

It could also be useful if the format specified a 'friendly' default camera position for the scene (this would also help with #510 as it would avoid the immediate need for the user to manipulate the camera when a scene loads) e.g. for the discovery playmat scenes - something like this:
[image]

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Thursday May 12, 2016 at 07:02 GMT_

----

Camera tag added in commit e9e04d8. I also changed the defaults in EnkiViewer to behave like in your screenshot.

Camera tag arguments are:
- `x` x-position of the camera
- `y` y-position of the camera
- `altitude` height of the camera
- `yaw` yaw rotation of the camera (0 == x-axis aligned, mathematical orientation when scene viewed from the top)
- `pitch` pitch rotation of the camera (0 == horizontal, positive is up, negative is down)

An example would be `<camera x="50" y="10" altitude="50" yaw=" 1.57" pitch="-0.8" />`

mbonani, 2019-02-12:
partial documentation is done by HowToBuildAPlayground.playground