Mobsya/aseba #21: Export as svg do not display the red dot in VPL

MobsyaBot, 2018-04-17:
[image] **Issue by [cbarraud](https://github.com/cbarraud)**
_Saturday Jul 30, 2016 at 13:00 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/540_

----

Hi

I found a small bug in the export of VPL as svg. When you export your program, it does not display the small red dot beside ground or prox sensors. In bmp or png it's perfectly ok. 

Here is an example:

[image]

On the left, a small program exported in png, on the right, on svg.

When you open the svg (I opened it with Inkscape), you can see that there is some sort of small dot at the place where the red one should be if you ungroup the image a few times in a row:

[image]

It seems that the degraded of red is not export correctly.

I work on a Windows 7 PC.

Here are the info on the Aseba version: 
[image]

Christophe

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Tuesday Aug 02, 2016 at 08:19 GMT_

----

It is indeed a bug in the SVG exporter of Qt (hence foreign issue). It seems that it is not able to export the circular gradient. I suggest we document this bug on the wiki and close it, because there is nothing we can do for now, excepted writing a minimal test case and reporting it to Qt.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Wednesday Dec 21, 2016 at 10:26 GMT_

----

Maybe Qt 5 will solve this problem, so I tag this issue 1.7

MobsyaBot, 2018-04-17:
[image] **Comment by [ypiguet-epfl](https://github.com/ypiguet-epfl)**
_Friday Feb 23, 2018 at 09:53 GMT_

----

Or just draw a plain red dot without gradient when you export to SVG.

mbonani, 2019-02-06:
works in QT5 and actual VPL

closed 2019-02-06 by mbonani