Mobsya/aseba #27: AsebaStudio : Répertoire temporaire persistant sous Windows

MobsyaBot, 2018-04-17:
[image] **Issue by [dburnier](https://github.com/dburnier)**
_Monday Dec 05, 2016 at 14:36 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/589_

----

Aseba Studio, du moins sous Windows, crée systématiquement un dossier aseba-studio-xxxx à chaque démarrage de l'application, contenant les fichiers *aseba-doc.qch* et *aseba-doc.qhc*.

Ces répertoires sont stockés sous C:\Users\%USER%\AppData\Local\Temp et prennent chacun 1,31 MBs. Cumulés dans le temps, cela dépassait le GB dans mon cas !!!

J'ai fait un test sous Windows avec la version 1.5.4-git-69c3165 et le problème est toujours présent.

Pour info, sous Linux, il y a bien un répertoire créé avec des fichiers dedans, mais il est détruit lorsque l'application AsebaStudio est fermée.

Merci de corriger le problème et de voir ce qu'il en est sous MAC os.

MobsyaBot, 2018-04-17:
[image] **Comment by [davidjsherman](https://github.com/davidjsherman)**
_Tuesday Dec 06, 2016 at 07:56 GMT_

----

Les outils Aseba peuvent utilement être mis à disposition sous forme de [Portable App](http://portableapps.com/about/what_is_a_portable_app), ce qui permettrait aux enseignants de les utiliser en classe sans installation permanent sur les postes de travail, mais à partir d'une clé par exemple. Il n'y aura pas de répertoire temporaire écrit sur le poste de travail et donc rien qui ne s'accumule.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Tuesday Dec 06, 2016 at 10:34 GMT_

----

This is probably because when the help file is deleted in `HelpViewer::~HelpViewer()` in line 69 of `client/studio/HelpViewer.cpp`, the object `helpEngine` is still alive, probably holding a reference to the help file, which prevents Windows from deleting it.

cor3ntin, 2019-03-06:
fixed by #427

closed 2019-03-06 by cor3ntin