Mobsya/aseba #32: Update codebase to C++14

MobsyaBot, 2018-04-17:
[image] **Issue by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Tuesday Apr 04, 2017 at 08:47 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/615_

----

The current Aseba codebase is mostly C++03. Recently, Aseba has been switched to C++14, and new code is expected to use the new C++14 features. Progressively, existing code should be migrated.

The following features should be used:

- **override**: methods that override parent ones should explicitly state that through the `override` keyword.
- **noexcept**: functions that are guaranteed not to return exception should state that through the `noexcept` keyword.
- **constexpr**: objects and functions that can be const expressions should state that through the `constexpr` keyword.
- **autogenm**: in interfaces with virtual destructors, when not manually specified, the following *auto-generated members* should be specified as `default`: copy constructor, copy assignment operator, move constructor, move assignment operator.
- **defval**: when it makes sense, class members should be given default values to simplify the constructors.
- **usingalias**: instead of `typedef`, keyword `using` should be for template aliasing.
- **enum**: use strongly-typed enums whenever meaningful.
- **nullptr**: use nullptr whenever a NULL pointer is used.
- **auto**: use auto and range-based loops whenever it makes sense.
- **contstit**: use `const_iterator` whenever possible, using `cbegin()` and `cend()`.
- **array**: use `std::array` instead of C arrays.
- **passbyvalue**: when a constructor copies a variable into its own, pass by value and `std::move`.

The following code parts of Aseba must be updated:

- [x] `common/`
- [x] `compiler/`
- [ ] `transport/dashel_plugins`
- [ ] `clients/studio`
- [ ] `clients/thymioupgrader`
- [ ] `clients/thymiownetconfig`
- [ ] other tools in `clients/`
- [ ] `switches`
- [ ] `targets`
- [ ] `examples`

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Wednesday Apr 05, 2017 at 15:03 GMT_

----

Commit 05d3b36710b0f09557925c0573873fecf8ed4aec updates `common/msg` to C++11 following these rules.

MobsyaBot, 2018-04-17:
[image] **Comment by [THS-on](https://github.com/THS-on)**
_Monday Apr 17, 2017 at 19:10 GMT_

----

Commit 05d3b36710b0f09557925c0573873fecf8ed4aec breaks build process because in tests/CMakeLists.txt the directory "msg" is defined, but is missing.  
If the line is removed building works fine.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Tuesday Apr 18, 2017 at 13:46 GMT_

----

Commit c4c5307 adds the missing files in 05d3b36710b0f09557925c0573873fecf8ed4aec, thank you for reporting the problem!

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Sep 22, 2017 at 09:44 GMT_

----

A large part of this can be done automatically, with supervision, using `clang-tidy`. With `FILENAME` as the cpp file to process, one can preview the changes with:
```clang-tidy -checks='modernize-*' -header-filter='.*' FILENAME -- -std=c++11```
and apply them with:
```clang-tidy -checks='modernize-*' -header-filter='.*' -fix FILENAME -- -std=c++11```

Additional include options should be given after `-std=c++11` when relevant.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Sep 22, 2017 at 09:45 GMT_

----

Commit 9802ba91 updates `compiler`.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Sep 22, 2017 at 09:53 GMT_

----

Commit 8b6e0f4e updates `common/utils`.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Sep 22, 2017 at 09:59 GMT_

----

Commit 80f6bae7 finishes updating `common/msg`.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Sep 22, 2017 at 10:10 GMT_

----

Commit daaf1be3 updates `common/zeroconf`.

MobsyaBot, 2018-04-17:
[image] **Comment by [cor3ntin](https://github.com/cor3ntin)**
_Thursday Feb 01, 2018 at 08:54 GMT_

----

Or you could use a compilation database

MobsyaBot, 2018-04-17:
[image] **Comment by [davidjsherman](https://github.com/davidjsherman)**
_Thursday Feb 01, 2018 at 09:17 GMT_

----

Is there a way to produce a compilation database that is compatible with Sonarqube? 
Currently we produce one using `build-wrapper` from [SonarCFamily for C/C++](https://docs.sonarqube.org/pages/viewpage.action?pageId=7996665) but it is sometimes a bit flaky and also closed-source.

MobsyaBot, 2018-04-17:
[image] **Comment by [cor3ntin](https://github.com/cor3ntin)**
_Thursday Feb 01, 2018 at 09:21 GMT_

----

@davidjsherman cmake does it, yes https://cmake.org/cmake/help/v3.5/variable/CMAKE_EXPORT_COMPILE_COMMANDS.html

MobsyaBot, 2018-04-17:
[image] **Comment by [davidjsherman](https://github.com/davidjsherman)**
_Thursday Feb 01, 2018 at 09:48 GMT_

----

@cor3ntin: Do you know whether they are compatible? If you can generate one, I can test it with sonar-scanner.

How do you recommend managing the three different compilation databases for the three different build environments?

cor3ntin, 2018-04-19:
Done in #139

closed 2018-04-19 by cor3ntin