Mobsya/aseba #34: disconection message should not be a windows

MobsyaBot, 2018-04-17:
[image] **Issue by [mbonani](https://github.com/mbonani)**
_Monday May 15, 2017 at 10:14 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/627_

----

When user unplug a Thymio it generate a windows that tell people to reconnect the robot and freezing the back windows. If they close it they cannot reconect without restarting aseba. 
We propose instead of a windows to add a banner that the robot was disconected and that aseba studio is trying to reconnect, without the possibility to stop that. People could then continue to use the main windows for example saving their code. 
The same should be implement ins VPL standalone

riedo, 2018-07-05:
I suggest to have a connection status directly in the IDE (see poorly rendered mock-ups below)

This reminds me of another question: could we have a reconnect button? what happens with the reconnection with the new switch? can the switch recognize the robot that was lost? what happens if the connection is lost but another robot is plugged?

[image] [image] [image]

cor3ntin, 2018-07-05:
> I suggest to have a connection status directly in the IDE (see poorly rendered mock-ups below)
That would be great

> This reminds me of another question: could we have a reconnect button?
Could be done, I'd prefer less buttons, more automatic behavior.

 > What happens with the reconnection with the new switch? can the switch recognize the robot that was lost? what happens if the connection is lost but another robot is plugged?

The new switch sends a notification when a robot connects or disconnects.
So the application can reconnect automatically.

However, natives ids cannot guarantee uniqueness (the id is often `1`), the solution would be to store a 128bits randomly generated id in the robot persistent memory. 

This can be done easily in thymio 3. For thymio 2, we need to look whether we have some bits to spare to do that (64 or 32 would be enough).

Then it would take a firmware update so the switch can read and set that id.

Once we have that, it would be easy for a client application to auto-reconnect to a robot with the same id.

Make sense?

cor3ntin, 2019-01-10:
Fixed

closed 2019-01-10 by cor3ntin