Mobsya/aseba #36: Serving local files from asebahttp and asebahttp2

MobsyaBot, 2018-04-17:
[image] **Issue by [ypiguet-epfl](https://github.com/ypiguet-epfl)**
_Monday Jun 19, 2017 at 14:49 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/642_

----

Suggestion: add support to asebahttp and asebahttp2 to also serve local files, to avoid cross-origin issues. A basic implementation with a single directory and a few mime types (html, js, css, png, jpg) would already be very useful.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Monday Jun 19, 2017 at 19:30 GMT_

----

Agreed, it makes sense.

MobsyaBot, 2018-04-17:
[image] **Comment by [davidjsherman](https://github.com/davidjsherman)**
_Monday Jun 19, 2017 at 20:52 GMT_

----

I'm not sure I understand. Could you describe a test that would demonstrate this? Some URIs and their expected results would help.

There is already a plan to serve the OAS apidoc and the thymio-motion.aesl file, I am guessing that this a more general service?

MobsyaBot, 2018-04-17:
[image] **Comment by [ypiguet-epfl](https://github.com/ypiguet-epfl)**
_Thursday Jun 22, 2017 at 07:02 GMT_

----

David, I'm thinking about web applications. Browsers usually restrict cross-origin requests with `XMLHttpRequest` (as I understand it, currently asebahttp2 is completely open with `Access-Control-Allow-Origin: *`). Another advantage is to avoid separate http servers or reverse proxys.

I'm not aware of development plans. Is there any link you can share?

MobsyaBot, 2018-04-17:
[image] **Comment by [davidjsherman](https://github.com/davidjsherman)**
_Thursday Jun 22, 2017 at 08:13 GMT_

----

I don't have my notes on the single-page app at the office, but can dig them up. They would be PDFs extracted from my livescribe. I am sure that they weren't made public in GitHub. Plans for a single-page app drove the definition of the new asebahttp API.

Historically the CORS modification in asebahttp was made in f45c83c038e03fe99a98a8b46e76eaeed2493bf1 in response to a discussion with Pierre Rouanet, but there was no issue opened for it. About that time there was a discussion about embedding the `thymio_motion.aesl` file in a firmware version of asebahttp. In part to avoid cross-origin but also to permit the Thymio to act as a wireless AP.

mbonani, 2018-08-23:
I suggest that the TDM can also put in the network file (improve .aesl) so that people can share code, share configuration, share exercice. It could be the way to make the "teachers tool".