Mobsya/aseba #45: language is not saved when you change it.

MobsyaBot, 2018-04-17:
[image] **Issue by [Vincebecker](https://github.com/Vincebecker)**
_Tuesday Oct 17, 2017 at 07:51 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/706_

----

If you switch the language from French to German (For example) Aseba will not remember it for the next time.

I know that Aseba use the language of the OS but it could be useful to redefine it.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Tuesday Oct 17, 2017 at 09:23 GMT_

----

Do you mean with Studio, using the new interface? Please when submitted bugs, indicate which version and platform you use, thank you ;)

I agree that it is a feature good to have?

cor3ntin, 2019-11-18:
Duplicates #459

closed 2019-11-18 by cor3ntin