Mobsya/aseba #46: Add support for refreshing txt field

MobsyaBot, 2018-04-17:
[image] **Issue by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Wednesday Nov 08, 2017 at 19:14 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/713_

----

The `zeroconf` class should provide a way to receive updates of the `TXT` field. This could be a method that goes through all detected targets and request as update the `TXT` field for each of them.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Wednesday Nov 15, 2017 at 16:34 GMT_

----

Commit 5326d03b46cd8e13bb9f7391e020e08d72b3559d introduces a member `detectedTargets` in `Zeroconf` that keeps track of detected targets until they are removed. This would allow to periodically refresh the `TXT` field using [DNSServiceQueryRecord](https://developer.apple.com/documentation/dnssd/1804747-dnsservicequeryrecord?language=objc) as discussed in more details in #690.

For now, we can simply deregister the busy services, and we can improve this in 1.7.

cor3ntin, 2018-08-22:
This issue is no longer relevant because of the thymio device manager

closed 2018-08-22 by cor3ntin