Mobsya/aseba #52: Add about box in firmware upgrader and network configurator

MobsyaBot, 2018-04-17:
[image] **Issue by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Dec 08, 2017 at 10:09 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/737_

----

Now that we have proper support for about box, it is easy. The following steps have to be done:
- add an about button and launch the about box
- link with qtabout
- initialize resource in main
- load translator
- add proper organization name for the sake of cleanness

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Dec 08, 2017 at 15:19 GMT_

----

This is not very critical so I am moving this issue to 1.7.

cor3ntin, 2018-08-22:
@mbonani do we still want that? I'm trying to get rid of the "thymio Launcher" milestone

cor3ntin, 2019-04-18:
The firmware upgrader will be removed ( feature fold back in the launcher)

closed 2019-04-18 by cor3ntin