Mobsya/aseba #54: Aseba Studio Help Out of date

MobsyaBot, 2018-04-17:
[image] **Issue by [paolorossetti](https://github.com/paolorossetti)**
_Tuesday Dec 12, 2017 at 08:41 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/741_

----

I am reading the help of Aseba but it refers to an old version of Aseba that no longer exists or I have a completely different one.
The interface is different and the behaviors described are not consistent with the user interface

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Tuesday Dec 12, 2017 at 09:18 GMT_

----

Yes this is a problem, thanks for writing it in an issue.

However, we won't have time to update that for 1.6. The plan is to migrate the source of the help from wikidot to the repository itself on github, which will greatly simplify the process of updating this help.

I have created a 1.6.1 milestone as I think that it is an important issue. We can do it before the switch to Qt5. @cor3ntin we should discuss that right after the 1.6 release.

cor3ntin, 2019-04-18:
The doc have been removed from studio altogether

closed 2019-04-18 by cor3ntin