Mobsya/aseba #56: Style cleaning

MobsyaBot, 2018-04-17:
[image] **Issue by [cor3ntin](https://github.com/cor3ntin)**
_Tuesday Dec 19, 2017 at 20:50 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/759_

----

Both `dashel` and `aseba` have some code styling inconsistencies that make it difficult to not introduce some ws-only change or styling issue.

Formatting can hopefully be automated away which leads to increased productivity in the long run.

I've made some experiment on dashel to that aim https://github.com/aseba-community/dashel/compare/master...cor3ntin:cleanws

-  Convert tab to space
`find \( -name "*.cpp" -o -name "*.h" \) -exec bash -x -c 'unexpand -t 4 "$0" | sponge "$0"' {}  \;`
- Remove trailing ws
`find \( -name "*.cpp" -o -name "*.h" \) -exec bash -x -c 'sed -i"s/[ \t]*$//" "$0"  ' {}  \;`

-Provide a custom clang format style that matches closely the style of `aseba`. That can be improved further.
https://github.com/cor3ntin/dashel/blob/eacaa85695ea433ab12ee800d5845530e0afd20b/.clang-format

Then it becomes very to ensure a consistent style.

The issue of course is that it introduces a big formatting change across of which it is painful to merge.
Timing the introduction of the change properly helps, otherwise conflict will have to be solve manually,
which I think is a small price to pay going forward.

Or we could only enforce the use of the tool on new files and diffs

https://clang.llvm.org/docs/ClangFormatStyleOptions.html
https://github.com/llvm-mirror/clang/blob/master/tools/clang-format/git-clang-format
https://dx13.co.uk/articles/2015/4/3/Setting-up-git-clang-format.html

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Wednesday Dec 20, 2017 at 06:58 GMT_

----

I am confused, you talk about changing tabs to space, but the clang style tells to use tabs. Is it only part of the cleanup process? If not, please do not force space, we have been using tabs for years because we like them.

For the rest yes it's good to enforce a consistent style.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Wednesday Dec 20, 2017 at 07:04 GMT_

----

Also regarding maintaining indentation on newlines: I am not convinced it is bad, it rather feels to me that the creator of git disliked it and decided to put it in red to force people into their convention. But if one gives more arguments on why it is better, I'm ready to reconsider my decision.

Does you style file keeps it or discards it?

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Wednesday Dec 20, 2017 at 07:07 GMT_

----

Note that this gives me a 404: https://github.com/asebacommunity/dashel/compare/master...cor3ntin:cleanws, probably it is private.

MobsyaBot, 2018-04-17:
[image] **Comment by [cor3ntin](https://github.com/cor3ntin)**
_Wednesday Dec 20, 2017 at 07:15 GMT_

----

Sorry, I meant space to tab.
Correct url is `https://github.com/aseba-community/dashel/compare/master...cor3ntin:cpleanws`

As for trailing whitespace at the beginning of a lines, most tools, including clang-format don't deal with them at all ( they forcefully remove them ), so the idea there is to make it easier for tools.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Wednesday Dec 20, 2017 at 10:39 GMT_

----

> Correct url is https://github.com/aseba-community/dashel/compare/master...cor3ntin:cpleanws

This page gives:
There isn’t anything to compare.
We couldn’t figure out how to compare these references, do they point to valid commits?

MobsyaBot, 2018-04-17:
> Correct url is aseba-community/dashel@master...cor3ntin:cpleanws

[image] **Comment by [cor3ntin](https://github.com/cor3ntin)**
_Wednesday Dec 20, 2017 at 10:44 GMT_

----

 Github keeps messing with the url `
https://github.com/aseba-community/dashel/compare/master...cor3ntin:cleanws`
- I wanted to avoid making a PR yet.

cor3ntin, 2018-04-19:
Fixed, a .clang-format has been merge

closed 2018-04-19 by cor3ntin