Mobsya/aseba #66: Dependencies Cleaning

MobsyaBot, 2018-04-17:
[image] **Issue by [cor3ntin](https://github.com/cor3ntin)**
_Wednesday Jan 24, 2018 at 17:32 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/797_

----

There is a number of dependencies we could get rid of or replace, given the occasion

- libxml2  => QtXml
- SDL2    => QtGamepad ( since Qt 5.7 )
- QWT    => QtCharts ( since Qt 5.7 )

MobsyaBot, 2018-04-17:
[image] **Comment by [davidjsherman](https://github.com/davidjsherman)**
_Wednesday Jan 24, 2018 at 17:36 GMT_

----

No, then that introduces a dependency on Qt that is not needed for many products. Asebahttp uses libxml2, Asebacompiler.js uses the SDL provided by Emscripten.

I feel like I am repeating myself, I said essentially the same thing in https://github.com/aseba-community/aseba/issues/793#issuecomment-360063972!

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Wednesday Jan 24, 2018 at 19:32 GMT_

----

I agree with David, libxml2 is a lower-level library that is well suited for embedded, and SDL2 is also much smaller than Qt and more suitable for command-line tools.

I'm fine with getting ride of Qwt if QtCharts brings similar features or better, but as it currently works well, I do not see the value of changing as there is non-trivial code rewrite to do. Do we have a comparison of the two feature-wise?

MobsyaBot, 2018-04-17:
[image] **Comment by [cor3ntin](https://github.com/cor3ntin)**
_Wednesday Jan 24, 2018 at 19:43 GMT_

----

My argument was that if Qt is already installed, QChart is more likely to be installed to. especially on Mac and Windows

I don't have a comparison but I think QChart would be a suitable replacement considering the use we have of it.  
https://doc.qt.io/qt-5.10/qtcharts-overview.html

MobsyaBot, 2018-04-17:
[image] **Comment by [davidjsherman](https://github.com/davidjsherman)**
_Wednesday Jan 24, 2018 at 20:18 GMT_

----

Marcel Dassault reputedly said that when designing a new version of an airplane, you change the flight surfaces, the motor, the weapons — but for each model, **change only one at a time**. My strong recommendation is that, if you must redo the build system, then do only that. Make it functionally equivalent to the old one, test it to make sure that there are no regressions, send it out for beta test and sit on it for a week.

If it works and is accepted, then it will be possible to leverage it to add new features.

MobsyaBot, 2018-04-17:
[image] **Comment by [cor3ntin](https://github.com/cor3ntin)**
_Wednesday Jan 24, 2018 at 20:24 GMT_

----

I was not suggesting to do that now.

cor3ntin, 2019-01-10:
We know use QtCharts instead of QWT and don't use libxml2 anymore

closed 2019-01-10 by cor3ntin