Mobsya/aseba #67: Aseba Studio looses connection with Aseba Playground on Mac

MobsyaBot, 2018-04-17:
[image] **Issue by [ypiguet-epfl](https://github.com/ypiguet-epfl)**
_Friday Jan 26, 2018 at 09:06 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/799_

----

Aseba Studio quickly looses the connection with Aseba Playground with Thymio.playground. This is observed on macOS 10.12 with 1.6, but maybe it's a more general problem. It happens after a short while, typically 10-60 seconds, 100% of the time. After the first disconnection, Aseba Studio automatically reconnects successfully and immediately disconnects in an infinite loop. The visible consequence is that the VPL window, if open, closes upon disconnection and reopens upon reconnection, and the list of variables etc. on the left of the main Studio window disappears and reappears.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Jan 26, 2018 at 09:23 GMT_

----

We observed that problem in the past as macOS is putting to sleep the background processes. We solved it in the release package by adding a flag to the Info.plist

Did you observe this problem with the released packages or your own compilation?

MobsyaBot, 2018-04-17:
[image] **Comment by [ypiguet-epfl](https://github.com/ypiguet-epfl)**
_Friday Jan 26, 2018 at 09:39 GMT_

----

With the released 1.6, without anything nonstandard (in Aseba Studio, About Aseba Studio, Libraries: Aseba version 1.6.0, build version git-c56c7b8d).

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Jan 26, 2018 at 11:11 GMT_

----

Interesting, maybe there is a bug in the packaging. Did the problem happen with the latest 1.5.x release ? @mbonani @cor3ntin was there any change in the packaging script that could have affected that?

MobsyaBot, 2018-04-17:
[image] **Comment by [ypiguet-epfl](https://github.com/ypiguet-epfl)**
_Friday Jan 26, 2018 at 11:22 GMT_

----

I've just tried with 1.5.5 and unifr.playground (thymio.playground is new in 1.6, I think). Same issue with the first simulated e-puck (without VPL obviously).

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Jan 26, 2018 at 11:24 GMT_

----

Ok, then we have to investigate, I put it in 1.6.1. Thank you for reporting and for the tests!

MobsyaBot, 2018-04-17:
[image] **Comment by [mbonani](https://github.com/mbonani)**
_Friday Jan 26, 2018 at 13:20 GMT_

----

Normally not, I see this behavior only when you minimize playground window. Perhaps we should not allow to minimze the windows.

MobsyaBot, 2018-04-17:
[image] **Comment by [ypiguet-epfl](https://github.com/ypiguet-epfl)**
_Friday Jan 26, 2018 at 13:47 GMT_

----

I haven't minimized the Playground window.

MobsyaBot, 2018-04-17:
[image] **Comment by [ypiguet-epfl](https://github.com/ypiguet-epfl)**
_Saturday Feb 03, 2018 at 08:41 GMT_

----

Poking Playground by clicking anywhere in its window will prevent disconnections for a short while. So it must be related to Playground having a snooze and Studio having a too short timeout. One of them should be changed; preferably Studio to avoid the same issue with other clients and because disconnections have really nasty consequences for the user.

MobsyaBot, 2018-04-17:
[image] **Comment by [mbonani](https://github.com/mbonani)**
_Wednesday Mar 07, 2018 at 09:13 GMT_

----

this could be probably solved with new .plist move to new CI

MobsyaBot, 2018-04-17:
[image] **Comment by [Zach1993](https://github.com/Zach1993)**
_Wednesday Mar 14, 2018 at 01:49 GMT_

----

Just started trying to use Studio and am having this problem. I have Aseba studio V 1.6.0 and OSX 10.13.3.

mbonani, 2019-06-11:
seems not to be still the case

closed 2019-06-11 by mbonani