Mobsya/aseba #68: Improve handling of SVG files

MobsyaBot, 2018-04-17:
[image] **Issue by [cor3ntin](https://github.com/cor3ntin)**
_Friday Jan 26, 2018 at 10:34 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/800_

----

In quite a few places, we use QSvg to render icons.
While this fine, it does introduce a hard dependency on QSvg. It would be possible to use QIcon instead, that would link to QSvg at runtime, removing that dependency

I think it's a better pattern that having modules such as `asebaaboutqt` be conditionally compiled.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Jan 26, 2018 at 11:14 GMT_

----

There was a reason I used QSvg directly but cannot remember the details, I think something linked to a problem with HiDPI screens or a limitation of QIcon with SVGs. It's fine for me to switch if the underlying problem has been fixed (which I expect to be the case as we switch from Qt 4.x to 5.x).

MobsyaBot, 2018-04-17:
[image] **Comment by [cor3ntin](https://github.com/cor3ntin)**
_Friday Jan 26, 2018 at 11:33 GMT_

----

@stephanemagnenat oh yeah, HiDPI support in 4.x was a mess - almost non existent really.
It's better now, i think we would still need to specify a default size in the svg files.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Jan 26, 2018 at 11:38 GMT_

----

> i think we would still need to specify a default size in the svg files.

That's annoying as we want to use the file at different resolutions. If this imply to keep using QSvg, I would keep using it.

MobsyaBot, 2018-04-17:
[image] **Comment by [cor3ntin](https://github.com/cor3ntin)**
_Friday Jan 26, 2018 at 11:56 GMT_

----

The default size is only useful if you don't specify it in the application.
It does not constrain the way you use the file, merely how viewers render
it. If the application request a given size, say 64x64, that's what it will
get

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Jan 26, 2018 at 13:08 GMT_

----

Ok then it's fine.

cor3ntin, 2019-08-28:
SVG is unconditionally required now

closed 2019-08-28 by cor3ntin