Mobsya/aseba #70: Aseba Studio crashes when it closes the VPL window during a drag

MobsyaBot, 2018-04-17:
[image] **Issue by [ypiguet-epfl](https://github.com/ypiguet-epfl)**
_Friday Feb 02, 2018 at 09:15 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/811_

----

When Aseba Studio closes the VPL window upon Thymio disconnection (see #799, but it probably also occurs upon more legitimate disconnections) and the user is dragging a VPL block, Aseba Studio crashes (real hard crash with `EXC_BAD_ACCESS`, not an uncaught c++ exception). Tested with 1.6 and Playground on macOS 10.12, but it's likely related to Qt and its use by Studio.

MobsyaBot, 2018-04-17:
[image] **Comment by [mbonani](https://github.com/mbonani)**
_Wednesday Mar 07, 2018 at 09:14 GMT_

----

this bug could wait the new CI

cor3ntin, 2018-10-31:
VPL has been completely extracted out of studio as a standalone app

closed 2018-10-31 by cor3ntin