Mobsya/aseba #74: MSVC Support

MobsyaBot, 2018-04-17:
[image] **Issue by [cor3ntin](https://github.com/cor3ntin)**
_Wednesday Feb 07, 2018 at 20:36 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/828_

----

We should probably make Aseba ( and dashel, enki, etc ) - compatible with the Microsoft toolchain, along `mingw64`

There are multiple reasons
 - Developers on windows are used to Visual Studio
 - Having more compilers tend to make for better, more portable code
 - Different compiler offer different warnings and diagnostics

The biggest effort is that dynamically linked symbols must be explicitly exported, though we could do hat on all platforms too. 
Otherwise there is probably not much work to do.

CI wise, we would need to install either `clang-cl` + `lld-link` or MSVC ( or both !). we would off course also need to install the windows sdk which we would have to think about in term of working with ansible but it's not too much of an issue either.
Cross compilation from linux is also possible ( but apparently we don't need it ).

I have set up some cross compilation toolchain files for my own use, I can share them if you want ( they do however not belong on github )

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Thursday Feb 08, 2018 at 07:09 GMT_

----

Dashel should compile on msvc out of the box. For information Tero from Qt Company managed to compile Thymio Adventure with msvc.
--
http://stephane.magnenat.net

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Thursday Feb 08, 2018 at 07:10 GMT_

----

and yes, more platform support is a good thing, also for debugging Windows problems.
--
http://stephane.magnenat.net

MobsyaBot, 2018-04-17:
[image] **Comment by [cor3ntin](https://github.com/cor3ntin)**
_Thursday Feb 08, 2018 at 08:20 GMT_

----

@stephanemagnenat dashel builds indeed ( with some warnings )  - but won't link because it does not export its symbols. I'll try to do a full static build of aseba later today to have a better understanding of the amount of work required.

MobsyaBot, 2018-04-17:
[image] **Comment by [cor3ntin](https://github.com/cor3ntin)**
_Thursday Feb 08, 2018 at 17:37 GMT_

----

Some work done here. No shared library support and I haven't tested the whole project yet https://github.com/aseba-community/aseba/pull/798/commits/9b0cf4cc78186a999cdaa908939ec161327bd4f2

cor3ntin, 2018-08-15:
MSVC is now the default windows compiler, support for mingw has been dropped, and clang-cl is also a supported target.
The VS version we use is v141

closed 2018-08-15 by cor3ntin