Mobsya/aseba #76: In Studio VPL, orphan about box upon reconnection

MobsyaBot, 2018-04-17:
[image] **Issue by [ypiguet-epfl](https://github.com/ypiguet-epfl)**
_Sunday Feb 11, 2018 at 16:24 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/831_

----

In Aseba Studio with the VPL window and its About dialog open, if the connection is broken and restored, the VPL window is reopened in front of the About dialog and has the focus while the About dialog cannot be closed. Clicking the info icon in the VPL window opens a second copy of the About dialog.

Steps to reproduce (tested with 1.6 on macOS 10.12):
- start Aseba Playground from a terminal in the foreground, i.e. don't end the command with `&`(easier to control the connection as explained below); on macOS, it could be something like `/Applications/Aseba/Simulations/Aseba\ Playground.app/Contents/MacOS/asebaplayground`
- in Aseba Playground, open thymio.playground
- open Aseba Studio (no need to do it in a terminal, just double-click its icon)
- connect to the simulated Thymio in Playground
- open the VPL window
- add a block to make sure that the VPL window will be restored upon reconnection (if empty, it isn't reopened)
- click the info icon (top right)
- in necessary, move it so that it cannot be completely hidden behind the VPL window once the VPL window will become frontmost
- in the terminal where you opened Aseba Playground, type control-z to suspend it
- wait a short while (a few seconds) so that Aseba Studio disconnects and closes the VPL window
- in the terminal, type `fg` to resume Aseba Playground in the foreground
- switch to Aseba Studio which reconnects automatically: the About dialog is behind the VPL window and cannot be moved or closed
- click the info icon: a second About dialog is opened and is active. The first About dialog remains open and inactive.

Suggestion: review completely modal windows in Studio. The VPL window shouldn't be closed without warning upon disconnection. A more partial fix would be to close the About dialog when the VPL window is closed.

Studio is fragile: shaking it like that easily crashes it.

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Monday Feb 12, 2018 at 08:02 GMT_

----

Thank you for reporting this bug. Your suggested "partial fix" is a good short term solution.

The reason why Studio closes the VPL window is that once the network is disconnected and reconnected, one cannot assume that the configuration is the same. Therefore, the design decision was to serialize and close all plugins upon disconnection, and restore them upon reconnection. I agree that for the user this is awkward when using VPL, especially for the standalone version, which is a hack over Studio. This design was created before VPL was introduced. For VPL2, we took a totally different approach, with VPL being the centre element and the connection being handle on the back. For the user it is much better.

mbonani, 2019-02-12:
in the new version studio as no pluggin anymore and there is only the standalone versions. Reconnection is managed by TDM and works also with dialobox opened.

closed 2019-02-12 by mbonani