Mobsya/aseba #9: Have the programs open directly if thymio is connected and you click the file

MobsyaBot, 2018-04-17:
[image] **Issue by [sharon13m](https://github.com/sharon13m)**
_Thursday Nov 14, 2013 at 17:33 GMT_
_Originally opened as https://github.com/aseba-community/aseba/issues/260_

----

Have the programs open directly if thymio is connected and you click the file.  Many times kids do this - no matter how  many times I tell them they need to open Aseba first.

MobsyaBot, 2018-04-17:
[image] **Comment by [mbonani](https://github.com/mbonani)**
_Thursday Nov 14, 2013 at 20:32 GMT_

----

That will be nice but consider also that .aesl file are used also for other robots and that they are target dependant.  what do you do when you hadn't the correctect robot connected? What do you do with playground? Several thymio?

MobsyaBot, 2018-04-17:
[image] **Comment by [sharon13m](https://github.com/sharon13m)**
_Thursday Nov 14, 2013 at 21:52 GMT_

----

A situation would be that someone downloads code and tries to double click to open it. Or that they save their file to desktop and double click to open.  *currently it always fails to open in program.. and student says.... it's not working!!"
Unfortunately I have not used other robots or multiple robots before.. so I'm unfamiliar with how it works exactly.  But can you have a logic that says ... if robot is there, connect program to robot.  If robot is not there, automatically open program in aseba.  And then as the other suggestion I made, be able to reconnect to robot within aseba without having to close and open program.
If aseba opens when .aesl  file is double clicked, if robot is connected you could automatically connect, or maybe show the other options that are available, and you can click to connect them in real time.  Having option to reconnect robot whenever needed is ideal.  A student should really only be opening and closing Aseba once per class period, currently there are multiple program restarts (whether it be for connection issues or bugs).
Does that make sense or am I missing something?

MobsyaBot, 2018-04-17:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Friday Nov 15, 2013 at 07:55 GMT_

----

A solution that I see is to make a helper program, that loads the aesl file, and look for which robot it is (based on its name, so that is just a heuristic), and proposes meaningful options:
- Thymio, suggest Thymio VPL, Studio, Playgroud 
- e-puck/Marxbot, suggest Studio, Playground
- other: load Studio automatically

cor3ntin, 2018-08-22:
We could do that in the launcher, I suppose.

mbonani, 2018-08-23:
yes the launcher could do that and it will be probably link to #36

mbonani, 2018-08-23:
aesl file should probably also contain configuration information or even perhaps exercice scenario. Having the Launcher that open by default aesl file can then help to dispatch exercice for in a school.
So from  a Launcher a teacher can expose on the net an aesl file that contain the course. TDM should also expose to the net (perhaps in an hidden way for normal user) the aels file at the end of the course so that teacher can collect all result. Probably sharing live the aesl (that are autosave) should help for the monitoring tool.

mbonani, 2019-02-12:
do we put that in 1.8?

cor3ntin, 2019-02-12:
Sure.
We probably need an UI to select either (Blockly, VPL, Scratch) or Studio, and a device if the preferred device is not connected