Mobsya/aseba #111: Block being dragged is shown below others event-actions pairs

MobsyaBot, 2018-04-18:
[image] **Issue by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Tuesday Jun 13, 2017 at 08:22 GMT_
_Originally opened as https://github.com/aseba-community/thymio-vpl2/issues/85_

----

As seen in this screenshot:
[image]

MobsyaBot, 2018-04-18:
[image] **Comment by [marvelous](https://github.com/marvelous)**
_Monday Jan 22, 2018 at 12:15 GMT_

----

The attached patch works except that the block jumps off the cursor when starting the drag. Ideas?
[85.txt](https://github.com/aseba-community/thymio-vpl2/files/1651843/85.txt)

MobsyaBot, 2018-04-18:
[image] **Comment by [stephanemagnenat](https://github.com/stephanemagnenat)**
_Monday Jan 22, 2018 at 12:26 GMT_

----

Some ideas:
- Could it be that the positioning is done relative to the parent, and that by changing parent the block jumps?
- Could it be something linked to `Drag.hotSpot`?
- Should one change anchoring as well as in http://doc.qt.io/qt-5/qtquick-draganddrop-example.html?

MobsyaBot, 2018-04-18:
[image] **Comment by [marvelous](https://github.com/marvelous)**
_Monday Jan 22, 2018 at 12:33 GMT_

----

- Positioning is done by QML's RowLayout, and setting it in a `PropertyChanges` changes nothing
- `Drag.hotSpot` is not set, and setting it changes nothing
- Anchors are not set, and setting them in an `AnchorChanges` changes nothing

mbonani, 2018-05-16:
this is still their I think

closed 2020-10-02 by mbonani