Mobsya/aseba #161: In asebahttp, replace objects with arrays of objects in json

ypiguet-epfl, 2018-06-18:
In asebahttp, node descriptions contain objects where the element order matters, e.g. for variables. In json, object properties are not ordered (see [http://json.org/](http://json.org/)).
Suggested fix: replace plain objects with arrays of objects where order matters.

cor3ntin, 2018-06-27:
What about having an actual index property? It seems to be a cleaner, less error prone solution

ypiguet-epfl, 2018-06-28:
An index used only for ordering? The only advantage would be backward compatibility, but since object properties are really unordered in json, we'd better choose the simplest type for ordered collections (arrays) and force clients to update. If you mean the actual value which depends on the order, such as the address of variables or the event id (see `HttpInterface::evNodes` in [aseba/switches/http/http.cpp](//github.com/Mobsya/aseba/blob/master/aseba/switches/http/http.cpp)), yes, that would be fine.

cor3ntin, 2018-06-28:
yep, I meant the address (or at least, some opaque thing that you can use to generate bytecode)

cor3ntin, 2018-06-29:
@ypiguet-epfl I'll do that in the new switch very soon, if you want it in the old switch, would you consider making a PR for it?

PR #243 (2018-09-20): Add index of aseba variables/events/functions
Fixes #161
commit 56e720fc (2018-09-20): Add index of aseba variables/events/functions

Fixes #161
closed 2018-09-20 by cor3ntin