Mobsya/aseba #213: Associating physical robots with there in-launcher icon

cor3ntin, 2018-08-21:
We want a way to light a given robot from the launcher so that each student can identify which robot it will use.
After thinking about it and talking with Vincent, here is the proposed design. 

In a classroom, there will be multiple launchers connected to the same set of robots. And if each launcher decides which color a robot can have, we may end up with 6 green robots, defeating our goal of identifying them.

Instead, deciding which color each robot should have will be left to the device manager

 - When a robot connects, it will be assigned a color by the thymio device manager, that color will remain assigned until the robot disconnects
- In the launcher, on rollover, we will send a message to the Thymio Device Manager to light the device
- The Thymio Device Manager will run some aseba code on the robot to light the robot of the given color
 - The Thymio Device Manager will inform the launcher that the robot is lit and will transmit the color information.
-  After a few seconds, the TDM will reset the thymio and inform the launcher that the robot isn't lit anymore.
- The launcher will use svg paths to match the color of the device.
- **Busy or locked robots can't be highlighted**

This ensure that the colors of the robot won't collide.
Vince will make a list of easily distinguisable colors for the TDM to choose from. 

Future direction:
 - Use a specific pattern
 - transmit the pattern frequency to the launcher ?
 - dedicated leds
 - sound :fearful: 

From a protocol standpoint
--> HighlightRobotRequest
<-- RobotHighlighted  #FF00FF
<-- RobotHighlighted  #000000

mbonani, 2020-05-07:
Friendly names are the first stet to distinguish robot, a flashing mechanism can be add in the robot menu as it show it works well. No colour are needed and only white, green or red are sufficient. I will propose at first step to only let the master computer (where the robot is really connect) to flash it.

mbonani, 2020-06-12:
think more about the user experience