Mobsya/aseba #300: Problem to build a Linux Package

CyrilleBiot, 2018-11-05:
Hi I'm trying to build aseba under linux (debian SID and Stretch)

But I've the same problem under these two distributions 

`cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=OFF ..`

```
CMake Error at aseba/clients/qtcommon/CMakeLists.txt:17 (target_link_libraries):
  Object library target "asebaqtcommon" may not link to anything.
```

```
-- Configuring incomplete, errors occurred!
See also "/home/serveur/compilation/git/mobsya/aseba/build/CMakeFiles/CMakeOutput.log".
See also "/home/serveur/compilation/git/mobsya/aseba/build/CMakeFiles/CMakeError.log".

```
And the CMakeError.log
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_c0877.dir/CheckFunctionExists.c.o  -o cmTC_c0877 -lpthreads
/usr/bin/ld : ne peut trouver -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_c0877.dir/build.make:97 : la recette pour la cible « cmTC_c0877 » a échouée
make[1]: *** [cmTC_c0877] Erreur 1
make[1] : on quitte le répertoire « /home/serveur/compilation/git/mobsya/aseba/build/CMakeFiles/CMakeTmp »
Makefile:126 : la recette pour la cible « cmTC_c0877/fast » a échouée
make: *** [cmTC_c0877/fast] Erreur 2
```

Any help is welcome

Thx
Cyrille

mbonani, 2018-11-05:
bonjour, vous essayer de compiler quel branche ?

CyrilleBiot, 2018-11-05:
Cool en français ;)
La branche master

CyrilleBiot, 2018-11-07:
Well, finally the cmke command works on SID but always the same errors on stretch yet I've the same packages installed on these 2 distributions (with the motif qt). Only these package is not installed on the Stretch : qt5-qmake-bin because it's not available in the stable repository.
? 
The log of these error: 
 ```
cat /home/serveur/compilation/git/mobsya/aseba/build/CMakeFiles/CMakeError.log
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/serveur/compilation/git/mobsya/aseba/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_72d23/fast"
/usr/bin/make -f CMakeFiles/cmTC_72d23.dir/build.make CMakeFiles/cmTC_72d23.dir/build
make[1] : on entre dans le répertoire « /home/serveur/compilation/git/mobsya/aseba/build/CMakeFiles/CMakeTmp »
Building C object CMakeFiles/cmTC_72d23.dir/CheckSymbolExists.c.o
/usr/bin/cc     -o CMakeFiles/cmTC_72d23.dir/CheckSymbolExists.c.o   -c /home/serveur/compilation/git/mobsya/aseba/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_72d23
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_72d23.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTC_72d23.dir/CheckSymbolExists.c.o  -o cmTC_72d23 
CMakeFiles/cmTC_72d23.dir/CheckSymbolExists.c.o : Dans la fonction « main » :
CheckSymbolExists.c:(.text+0x1b) : référence indéfinie vers « pthread_create »
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_72d23.dir/build.make:97 : la recette pour la cible « cmTC_72d23 » a échouée
make[1]: *** [cmTC_72d23] Erreur 1
make[1] : on quitte le répertoire « /home/serveur/compilation/git/mobsya/aseba/build/CMakeFiles/CMakeTmp »
Makefile:126 : la recette pour la cible « cmTC_72d23/fast » a échouée
make: *** [cmTC_72d23/fast] Erreur 2

File /home/serveur/compilation/git/mobsya/aseba/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/serveur/compilation/git/mobsya/aseba/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_647da/fast"
/usr/bin/make -f CMakeFiles/cmTC_647da.dir/build.make CMakeFiles/cmTC_647da.dir/build
make[1] : on entre dans le répertoire « /home/serveur/compilation/git/mobsya/aseba/build/CMakeFiles/CMakeTmp »
Building C object CMakeFiles/cmTC_647da.dir/CheckFunctionExists.c.o
/usr/bin/cc    -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_647da.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.7/Modules/CheckFunctionExists.c
Linking C executable cmTC_647da
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_647da.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_647da.dir/CheckFunctionExists.c.o  -o cmTC_647da -lpthreads 
/usr/bin/ld : ne peut trouver -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_647da.dir/build.make:97 : la recette pour la cible « cmTC_647da » a échouée
make[1]: *** [cmTC_647da] Erreur 1
make[1] : on quitte le répertoire « /home/serveur/compilation/git/mobsya/aseba/build/CMakeFiles/CMakeTmp »
Makefile:126 : la recette pour la cible « cmTC_647da/fast » a échouée
make: *** [cmTC_647da/fast] Erreur 2

```

mbonani, 2018-11-07:
pour info la branche master est en grand chanboulement donc ce tout ne fonctione pas, mais par contre elle devrait compiler, voire notre travis.
C'est quoi votre but?

CyrilleBiot, 2018-11-07:
Je cherche simplement un paquet debian d'aseba (non ubuntu, mais en sid et stretch comme le sont les machines de mon école)
J'avais donc recompilé le tout depuis aseba-community (pour avoir la dernière version d'aseba ; ça marche, ils tournent nickel) mais sur le forum de Thymio on m'a conseillé de plutot utiliser le git de mobsya, donc j'essayais avec ce repo. 
Tout simplement.
Ensuite ceux packagés via aseba-community sont fonctionnels, donc pas de réelle urgence ;)
Surtout de la curiosité ;)

mbonani, 2018-11-08:
J'écris en anglais pour que se soit util à d'autre. 

So actual situation on nov. 2018 is that mobsya and aseba comunity fork the development of aseba. Mobsya concentrate on Thymio.

- Master branch (based on qt5) of Mobsya is not stable and lot of change appen to fullfill https://github.com/Mobsya/aseba/projects/3 . The last release is on branch release-1.6.x (based on qt 4.8) where we release packet for windows and macosx 1.6.1.

- On aseba-comunity the master branch was pass on qt 5 and is near 1.6.1 of  mobsya. 

So for making packet for debian it is more stable on mobsya release 1.6.x but it is in qt4.8.

cor3ntin, 2018-11-08:
The master branch requires CMake 3.12+.
But, as @mbonani said, we don't recommend using the master branch at this point

closed 2018-11-08 by cor3ntin

CyrilleBiot, 2018-11-09:
OK thx. I'll try this asap
++