Mobsya/aseba #311: Communication error during upgrade firmware 12 on Thymio (aseba 1.6)

raypel78, 2018-11-16:
Hello
On one robot when I try to upgrade from firmware 10 to the firmware 12, I have the following error

_Une erreur de communication durant le processus de mise à niveau : I/O error (2) : cannot write to file (max retry reached). Le fichier spécifié est introuvable._

To turn around the problem I used the asebacmd mode line 

asebacmd -t "ser:name=Thymio" wusb 1 Thymio2-V10.hex reset

I have the following result

_Found Thymio on port \\.\COM18
Flashing Thymio2-V10.hex
Entering bootloader
In bootloader, about to write 65 pages
Writing page 1... Waiting ack ... Success
Writing page 2... Waiting ack ... Success
.....
Writing **page 60**... Waiting ack ... Success
Writing **page 82**... Waiting ack ... Failure
Warning, error while writing page 82, continuing ...
Writing page 83... Waiting ack ... Failure
Warning, error while writing page 83, continuing ...
Writing page 84... Waiting ack ... Failure
Warning, error while writing page 84, continuing ...
Writing page 85... Waiting ack ... Failure
Warning, error while writing page 85, continuing ...
Writing page 0... Waiting ack ... Success
Write completed
Exiting bootloader_

After that the robot is operational with firmware version 10

Thank for your answer

Raymond

mbonani, 2018-11-20:
Hi,
I suppose you use the Thymio Firmware Upgrader, sometimes the .hex files that is download is not correct and makes this kind of error.  Normally relaunching it, solve the problem.
Then what you get using asebacmd is correct, if you want firmware 12, just download it and replace the command line with` asebacmd -t "ser:name=Thymio" wusb 1 Thymio2-V12.hex reset`
It is also normal that the 4 last page cannot be written, it is the bootloader place and it protect itself. We just give a full .hex that can also be upload in production with a pickit

raypel78, 2018-11-22:
Thank you for your answer,
I tried two times with firmware upgrader and I have the same problem 2 times (on the same robot). .
For my others robots, if Firmware Upgrader does not work I will use command line.

closed 2018-11-22 by raypel78