Mobsya/aseba #340: load run in studio 1.6 and play in 1.7 with Wireless Thymio

mbonani, 2018-12-05:
I made some test with firmware 12 and 4 Thymio on the same key. I load a big code that is 99% of the bytecode space.  It seems that the bytecode is update correctly one robot by one but then the status is sometimes not passing:
with 1.6 if you press to quickly run then some robot not start
with 1.7 the bytecode are update but then robot are in paused and not running, specially if the sync is active.
Probably we can be improve this as it seems that only the status is not passing well.

cor3ntin, 2019-01-09:
That should be fixed

cor3ntin, 2019-01-09:
@mbonani can i let you test that ?

mbonani, 2019-02-12:
it seems better, the button for all are just confusing because naming are not the same

closed 2019-03-06 by cor3ntin