Mobsya/aseba #380: multiple tabs in different networks not allowed

ogoletti, 2019-02-07:
when programming two different robots on two different networks through two different dongles, one has to open two different studios. Is it expected ? I suggest it would be more clear to allow multiple tabs in one studio (eventually of different colors one/network?). 
On macos this is difficult because launching a second time the studio has to be done through command-line with `open -n _app_`. 
Moreover, there is no information in the studio to add a thymio nor any indication on why the second thymio (on another network) can't be connected, even though at startup the two thymio are visible as targets.

cor3ntin, 2019-02-07:
> when programming two different robots on two different networks through two different dongles, one has to open two different studios. Is it expected?

Yes.
Some features like saving assume all robots are in the same network

> On macos this is difficult because launching a second time the studio has to be done through command-line with `open -n _app_`.

This will be fixed by the launcher in the upcoming version 

> Moreover, there is no information in the studio to add a thymio nor any indication on why the second thymio (on another network) can't be connected, even though at startup the two thymio are visible as targets.

In 1.7, studio is launched through a launcher so it will be easier to understand what robots are in the same networks and launching studio on for the robots you want will be simple

mbonani, 2019-02-07:
Actually aseba studio 1.6 is suppose to connect to a unique network. You can put robots on same network by pairing the Thymio to the same dongle. You will get then the two tabs in the same asebastudio. Thymios will also then exchange messages  also without aseba running. But actually using lot of robot on same network will probably show you some problem that are mostly fix inaseba 1.7 . we are improving also the connection mechanism.

mbonani, 2019-02-07:
Then combining different network will be made in a second step