Mobsya/aseba #381: opening .aesl file clears other tab

ogoletti, 2019-02-07:
I don't know if this issue is linked to  #288 (which I also experienced). 

When you program two thymio on the same network in two different tabs. When you load a .aesl file in one tab, then the other tab is cleared. Here is my workflow (to avoid  #288) : 
0) connect one dongle and two thymio on the same network
1) open studio 
2) program first thymio in first tab and save.
3) program second thymio in second tab and save.
4) close studio
5) open studio
6) load either file (it loads in the correct tab)
7) load the other file (it loads in the correct tab) at this point, the other tab has been cleared.

cor3ntin, 2019-02-07:
What version of aseba are you using ?

ogoletti, 2019-02-07:
1.6.0

mbonani, 2019-02-07:
#288 is for 1.7
what you are saying seems me normal when you save a file it will normally save it for all the tabs (entire network present) so if you program only one tabs it will save that the second is empty. So you have to program all you network at the same time.

closed 2019-02-07 by mbonani