Mobsya/aseba #404: closing Launcher may crash studio

mbonani, 2019-02-13:
When you have a studio running , if you close Launcher it makes crash studio. It is specially doing that when you make it a second time. 
First time you close studio is working showing that robots are unconnected you relaunch and robot connects again. Closing launcher second time crashes studio.

cor3ntin, 2019-02-13:
crap...

cor3ntin, 2019-03-04:
Ok - on linux - maybe windows specific ?

cor3ntin, 2019-03-05:
@cbarraud did you repro this one ?

mbonani, 2019-03-11:
still there, not systematic but it happen after 2-3 trials

 Exception non gérée à 0x00007FFA9D3C14EE (Qt5Core.dll) dans asebastudio.exe : 0xC0000005 : Violation d'accès lors de la lecture de l'emplacement 0xFFFFFFFFFFFFFFFF.
[image]

cor3ntin, 2019-03-11:
What is the reproduction method exactly ?

mbonani, 2019-03-11:
open launcher and studio
close only the launcher and open it again until it make crach studio on a close of the launcher

PR #444 (2019-03-13): Fix crash of qt apps when a TDM is disconnected.
QObjects should not be deleted directly.

Fixes #440
Fixes #404
commit 58dff9f9 (2019-03-13): Fix crash of qt apps when a TDM is disconnected.

QObjects should not be deleted directly.

Fixes #440
Fixes #404
closed 2019-03-13 by cor3ntin