Mobsya/aseba #415: TDM shoul reconnect to firmware 12 robots, if possible?

mbonani, 2019-02-21:


cor3ntin, 2019-02-22:
I'm not sure I remember which firmware 12 is.
Re-Connection (as in "oh, I know this device") absolutely needs the UUID

mbonani, 2019-02-22:
12 is without UUID, 
So perhaps we should: improve the upgrade facilities or good communicate on it.

cor3ntin, 2019-02-22:
Sure, that is planned for later, in the meantime I think we can hardcode a message in the launcher

mbonani, 2019-03-01:
simple test if <13 show upgrade message

cor3ntin, 2019-03-12:
#438

- Added simple icon + tooltip to notify user of tooltip
- The TDM will check update when plugging a device
- The TDM now therefore has the machinery to do https - which is what took the most time
- Device with fw or older are hardcoded to have an update

closed 2019-03-12 by cor3ntin