Mobsya/aseba #419: improve clap block in VPL by avoiding self-generated music

riedo, 2019-03-01:
many people complain that clap event" music action = buzzing noise.
We could change the code for clap event and music action to: (see lines concerning sound_is_playing)

(clap event)
```
# setup threshold for detecting claps
mic.threshold = 250

onevent mic
        if !sound_is_playing then
              #whatever code
        end
```
(music action)
```
# sound playing flag
var sound_is_playing=0
# variables for notes
var notes[6]
var durations[6]
var note_index = 6
var note_count = 6
var wave[142]
var i
var wave_phase
var wave_intensity

# compute a sinus wave for sound
for i in 0:141 do
	wave_phase = (i-70)*468
	call math.cos(wave_intensity, wave_phase)
	wave[i] = wave_intensity/256
end
call sound.wave(wave)

# when a note is finished, play the next note
onevent sound.finished
	if note_index != note_count then
		call sound.freq(notes[note_index], durations[note_index])
		note_index += 1
        else
                sound_is_playing = 0
	end

onevent Whatever
	call math.copy(notes[0:5], [whatever notes])
	call math.copy(durations[0:5], [whatever durations])
	note_index = 1
	note_count = 6
	call sound.freq(notes[0], durations[0])
	sound_is_playing=1
```

mbonani, 2019-03-04:
I agree it can help current user. we were more to not change anymore VPL, but I can agree it can be done in this case.  @cor3ntin is it easy to implement?

cor3ntin, 2019-03-04:
Again, i have no idea...