Mobsya/aseba #468: Possible to zoom the border of Scratch interface?

cbarraud, 2019-03-28:
I was wondering if there was a way to zoom in the border of the Scratch interface and not just the "coding area". Here is what I have when I start it:

[image]

And this is what I have when I zoom the "coding area":

[image]

But I couldn't find a way to zoom the border region. The text in the side blocks can be quite small there.

cor3ntin, 2019-03-28:
@cbarraud i think you might have better luck asking that upstream https://github.com/llk/scratch-gui
It will be difficult for us to make modification to the scratch engine and the more we do the less we will be able to adopt new scratch versions

mbonani, 2019-03-28:
[image]
dans scratch3 online s ctr+roulette sur la barre bleu fait le zoom sur tout la page... est-ce que c'est pas qulque chose à activé dans Chromium?

cor3ntin, 2019-03-28:
@mbonani that we probably can do indeed

mbonani, 2021-12-13:
seems to work

closed 2021-12-13 by mbonani