Mobsya/aseba #496: Uniformise names Thymio Suite 2.0

cbarraud, 2019-04-12:
In some part of the app, the names are not consistent. We should have "Thymio Suite 2.0" or "Thymio Suite" everywhere.

Names of the packages to download should be "Thymio-suite-2-0"
[image]

The window of Thymio-Launcher should be called "Thymio Suite"
[image]

The name of the app in the repository of the OS should be "Thymio Suite"
[image]

In the installer, we should have "Thymio Suite 2.0" (everywhere in the wizard)
[image]

closed 2019-04-15 by cor3ntin

cbarraud, 2019-04-16:
I found here that the name of the app is still "Thymio launcher.exe"
[image]

Here is the version (just downloaded now)
[image]

I have to search for Thymio launcher then instead of Thymio Suite.
[image]

I tested on Windows 8.1.

mbonani, 2019-04-23:
please try to uninstall first your old version

PR #500 (2019-04-15): Use the name Thymio Suite consistantly for the launcher and installer
Fixes #496