Mobsya/aseba #498: Building on Fedora30 fails

nielsk, 2019-04-13:
enki and aseba cmakes tell me that it cannot find Boost
```
-- The following packages have not been found:

 * Boost (required version >= 1.67)

```
But boost is installed:
```
boost-devel.x86_64                                1.69.0-6.fc30                             @fedora                           
boost-python2-devel.x86_64                        1.69.0-6.fc30                             @fedora                           
boost-python3-devel.x86_64                        1.69.0-6.fc30                             @fedora     
```
enki builds though.

When I try to build aseba, it errors out with this error:
```
[ 42%] Automatic MOC for target asebastudio
[ 42%] Built target asebastudio_autogen
[ 42%] Building CXX object aseba/clients/studio/CMakeFiles/asebastudio.dir/VariablesModel.cpp.o
/home/nik/src/aseba/aseba/aseba/clients/studio/VariablesModel.cpp: In lambda function:
/home/nik/src/aseba/aseba/aseba/clients/studio/VariablesModel.cpp:75:76: error: lambda parameter ‘key’ previously declared as a capture
   75 |                                    [&key](const auto& ptr, const QVariant& key) { return lessThan(ptr->key, key); });
      |                                                            ~~~~~~~~~~~~~~~~^~~
make[2]: *** [aseba/clients/studio/CMakeFiles/asebastudio.dir/build.make:284: aseba/clients/studio/CMakeFiles/asebastudio.dir/VariablesModel.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:10985: aseba/clients/studio/CMakeFiles/asebastudio.dir/all] Error 2

```

nielsk, 2019-04-13:
I found now other build-instructions: https://github.com/Mobsya/aseba/blob/master/docs/en/development/building.rst
I will try now those…

nielsk, 2019-04-13:
Still errors out:
```
[ 46%] Building CXX object aseba/clients/studio/CMakeFiles/asebastudio.dir/VariablesModel.cpp.o
/home/nik/src/aseba/aseba/clients/studio/VariablesModel.cpp: In lambda function:
/home/nik/src/aseba/aseba/clients/studio/VariablesModel.cpp:75:76: error: lambda parameter ‘key’ previously declared as a capture
   75 |                                    [&key](const auto& ptr, const QVariant& key) { return lessThan(ptr->key, key); });
      |                                                            ~~~~~~~~~~~~~~~~^~~
make[2]: *** [aseba/clients/studio/CMakeFiles/asebastudio.dir/build.make:284: aseba/clients/studio/CMakeFiles/asebastudio.dir/VariablesModel.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:10985: aseba/clients/studio/CMakeFiles/asebastudio.dir/all] Error 2
make: *** [Makefile:163: all] Error 2

```

cor3ntin, 2019-04-13:
The fact that this ever compiled is beyond me.
Sorry about that. 
Do you know which version of GCC you use ?

cor3ntin, 2019-04-13:
Let me know if you still have issues

closed 2019-04-13 by cor3ntin

nielsk, 2019-04-13:
Compiling now. It goes beyond that error already :)
gcc version: gcc (GCC) 9.0.1 20190312 (Red Hat 9.0.1-0.10)

nielsk, 2019-04-13:
It compiled now but it is more or less unusable apparently (no .desktop-file for thymiovplclassic apparently; the dongle gets detected by the wireless-tool but thymiovplclassic won't connect, asebastudio won't create new files etc).
But that would be something for other tickets.

mbonani, 2019-04-13:
All this tools are now managed by the Thymio launcher (Thymio suite) have
you try the flatpak?
There is doc inside the repo

cor3ntin, 2019-04-13:
@nielsk try launching `thymio-launcher`

nielsk, 2019-04-13:
thymio-launcher just stays blank and that's what I see in the cli:
```
thymio-device-manager starting
qrc:/qml/main.qml:2:1: module "QtQuick.Controls" version 2.4 is not installed
thymio-device-manager started

```
I already disabled selinux.
I will search now for the flatpak. I couldn't find one on flathub

nielsk, 2019-04-13:
It works with the flatpak. Yeah. Now to figure out how to remove the compiled stuff and how to transfer the already created files from the VM.

cor3ntin, 2019-04-13:
You are missing the `qt5-qtquickcontrols` package, probably

nielsk, 2019-04-13:
I installed it; didn't help.
The flatpak works splendidly though :)

cor3ntin, 2019-04-13:
Hum. It might be `qt5-qtquickcontrols2` ( I don't have fedora on hand )
But if you are just trying to run the application, the flatpak is definitively the recommended solution.

Being on flathub is definitively something we will strive to do in the near future.

PR #499 (2019-04-13): Fix build on fedora
fixes #498