Mobsya/aseba #503: Scratch messages seem to not all be transmitted

cbarraud, 2019-04-17:
I used Scratch 3 with Windows 8.1 and Thymio Suite:
[image]
I did this program:
[image]
and pressed the different buttons, and I have the following behaviours:

1. The robot sometimes doesn't react to the pression of a button even though you see that the button is pressed (sound + light on the button)
2. Sometimes, when you use the block "all motor speed 200", one of the motor start and not the other one or inversely if you want to stop both motors (only one stops)
3. I sometimes get lag between an action and a reaction (like 100 ms)

I tried with Wireless Thymio and classic Thymio, Firmware 13. I see this behaviour in all cases, maybe a bit more with the Wireless one

cbarraud, 2019-04-17:
By the way, one of our partner asked me that 20 minutes ago :) :
"In addition, do you plan to offer a way to download Scratch 3 code onto the robot so it can work offline with code from scratch? This can be super useful for customers and will solve our lagging issues and scratch bugs where some commands are being skipped over while executing... "

Would it be possible?

mbonani, 2019-04-23:
For the first one it is strange that it is appening with a cable robot...
The second one is no.

mbonani, 2019-04-24:
your test work fine for me but I add lot of problem with this simple code
[image]
also with cable robot sometime a motor do not get its normal value

mbonani, 2019-06-04:
there was two sprites with code doing similar things  so it was a confusing test

cor3ntin, 2019-06-04:
@mbonani how often, with what code ?

mbonani, 2019-06-04:
in Wireless mode for me its appens 2-3% of the time that one of the motor do not get a message, it is more difficult to say when both have not react to the commend it is the button that was not correctly transmit or the two motor do not get the message.

 for me we should avoid a block sending several messages. one block= one message

cor3ntin, 2019-06-04:
Other solutions would be to send one aseba messages for multiple events

mbonani, 2019-06-04:
in which sense? getting the value are normally manage by the TDM, I think it make sense to get the changed value, with the new protocol but we could be more inteligent for specific events. In the other sense one scratch blocks should send an unique aseba "user" message that acts on something. If we can directly act on a variable it just send an unique "systeme message"

I just analyse more the .aesl and we can improve:

- calculation for the proximity sensor should be done under `onevent prox` [10Hz] and not `event button` [50hz] it will relaxe the VM, 
- use the math.deque (http://wiki.thymio.org/en:asebastdnativedeque) only available from firmware V11
- RGB leds as variables? (firmware V13) can we send a system message to change 3 variables or do we pass as now with user message?
- use the `onevent button.backward` and related to emit a message to say a button was pressed or relaxed, I think we missed some pressed button because TDM read at 10hz the variables and buttons are manged at 50Hz internaly so a very short pressed can be missed (probably what is happening to Christophe)

mbonani, 2019-06-06:
most sugestion implemented in PR https://github.com/Mobsya/scratch-vm/pull/40

cor3ntin, 2019-06-06:
Fixed ?

mbonani, 2019-06-06:
let me test inside master :-)

mbonani, 2019-06-07:
New version improve and fix the case for the buttons problems on cable normaly it cannot miss one.
But there is still the problem for the Wireless robot where some "user" message can be drop without preventing anything.  So sometimes a button can be push, the code emit a sound the vm send the message, but the computer do not get it in respect to the Wireless aviability.

mbonani, 2019-06-07:
It seems to pose sometimes problem to q-motion behavior, I remember that the old code was managing that but I am not sure i is still the case

mbonani, 2019-06-07:
I think I will open specific issue in scratch repo and close this one

mbonani, 2019-06-07:
closing issue because reported in:
https://github.com/Mobsya/scratch-vm/issues/42
https://github.com/Mobsya/scratch-vm/issues/41

closed 2019-06-07 by mbonani