Mobsya/aseba #530: graph in studio should have stop and clear button and export options

mbonani, 2019-05-06:
[image]

mbonani, 2020-02-20:
- [ ] add checkbox for time sliding windows 
- [ ] add time input of sliding windows 
- [ ] add pause button
- [ ] add export feature

actual refresh button is on top, re-think where to put all this feature

ValentinaTibs, 2020-04-03:
What  have to happens when the pause button is pressed?

mbonani, 2020-04-03:
it has to freeze the graph at the actual view

ValentinaTibs, 2020-04-07:
and what happens to the X axe when the graph is unpaused?

ValentinaTibs, 2020-04-15:
[ ] removing the custom dark enable palette that differs from theme to theme
[ ] removing the unity after the 0 from the X-Axis and adding a blank every 3 digits 
[ ] adding the unit measure (milliseconds) in both X-axes and input box
[ ] fix the size of the input box to 6 digit

closed 2020-04-16 by ValentinaTibs