Mobsya/aseba #533: update text explanation of each programm and tranlate them

mbonani, 2019-05-06:
[image]

- Title are in  double
- Links as to be update to the new site
- do we need other infomations informations ?
-----
Mandatory 
- [ ] fr
- [ ] en
- [ ] it
- [ ] de
-----
Optional
- [ ] pl
- [ ] es
- [ ] el
- [ ] ja
- [ ] ru
- [ ] tr
- [ ] zh

cbarraud, 2019-07-29:
@cor3ntin ca you see for german

cor3ntin, 2019-08-16:
@cbarraud done

mbonani, 2019-09-11:
there is still something wrong with the css

mbonani, 2019-09-11:
css is fix in https://github.com/Mobsya/aseba/pull/650

mbonani, 2019-10-17:
I remove from the  milestone because mandatory language are done

mbonani, 2020-02-14:
translation is fully completed alos for pl

mbonani, 2020-05-07:
some pl are missing, other language are not covered see to clarify the way to do it and expose to interested people.

mbonani, 2020-06-02:
pl is up to date in version 2.0.3

mbonani, 2021-12-14:
FR IT DE EN up to date for 2.3.0