Mobsya/aseba #541: [Linux] Nothing in the Thymio windows

MrCrapaud, 2019-05-12:
Hello all,

I just tried to install Thymio under Linux (Kubuntu 18.04, installed only for my daughter so that she can use Thymio! :-)). I didn't encounter any problem during the installation, but when I launch Thymio through the "flatpak run org.mobsya.ThymioSuite" command, I only get an empty window (with a Thymio icon in the upper-left corner of the window). In the terminal, I get the following messages:

_Searching for  "thymio-device-manager"
"thymio-device-manager" found:  "/app/bin/thymio-device-manager"
thymio-device-manager starting
"fr_FR" "launcher_" _":/translations"
"fr_FR" "qt_" "/usr/translations"
Didn't load translation "qt_"
qrc:/qml/main.qml: File was compiled ahead of time with an incompatible version of Qt and the original file cannot be found. Please recompile
thymio-device-manager started__

In fact I see more lines displayed but since they seem less relevant to me and contain information about my user name and host name, I'd prefer to start only with this (please tell me if I can send your more through a private message!). Do you have any idea of what my problem is? Is it linked to Qt?

Thanks in advance for your support!

Best Regards,
L.

mbonani, 2019-05-13:
thank for the feedback

loicmathieu, 2019-05-22:
Hello,

I encounter the same issue on Ubuntu 19.04

```
flatpak run org.mobsya.ThymioSuite
Gtk-Message: 14:04:31.087: Failed to load module "canberra-gtk-module"
Gtk-Message: 14:04:31.087: Failed to load module "canberra-gtk-module"
Qt: Session management error: None of the authentication protocols specified are supported
Searching for  "thymio-device-manager"
"thymio-device-manager" found:  "/app/bin/thymio-device-manager"
thymio-device-manager starting
"fr_FR" "launcher_" ":/translations"
"fr_FR" "qt_" "/usr/translations"
Didn't load translation "qt_"
qrc:/qml/main.qml: File was compiled ahead of time with an incompatible version of Qt and the original file cannot be found. Please recompile
thymio-device-manager started
```

Regards,

Loïc

cor3ntin, 2019-05-28:
Should be fixed on master pending rebuild. Thanks for reporting !

closed 2019-05-28 by cor3ntin