Mobsya/aseba #602: aesl files are not compatible between aseba 1.6.1 and thymio suite 2.0

riedo, 2019-08-16:
A user reported that aesl files created in Thymio Suite do not open in Aseba 1.6.1.
In order to test I created the same code in both versions and tried to open them in both versions.

The 1.6.1 file opens in 2.0, but no code appears in the window. The 2.0 file won't open in 1.6.1

comparison of both files (2.0 on the left, 1.6.1 on the right)
[image]
:
Why change the file format?

mbonani, 2019-08-30:
I remember we add fix that in the past... to be checked.

cor3ntin, 2019-09-05:
> The 1.6.1 file opens in 2.0, but no code appears in the window. 

I can't reproduce this issue - can you confirm you have it ? Can you send the files

riedo, 2019-09-11:
testing today, the file created in 1.6.1 opened correctly in 2.0. The files created in 2.0 opens in 1.6.1 but says the correct node was not found, so it opens in a different tab.
[toto.zip](https://github.com/Mobsya/aseba/files/3601175/toto.zip)

closed 2019-09-11 by mbonani