Mobsya/aseba #604: in launcher application description is not shown in correct language for macosx

mbonani, 2019-08-20:
probably missing files?

mbonani, 2019-09-05:
this append on my macs for testing, OS is in english, description of programs are in english but programs and Suite are in french. On mac OS 10.12.6 Sierra
[image]

mbonani, 2019-09-05:
I put OS in french all seems ok
[image]

mbonani, 2019-09-05:
I reput english and all seems fine 
[image]

mbonani, 2019-09-05:
mismatch is when you have both language install on your mac 
[image]

cor3ntin, 2019-09-05:
Seems related to that https://bugreports.qt.io/browse/QTBUG-72491

mbonani, 2019-09-06:
just to document but for me it is not a priority, whats append, if "primary" language is different of english all program is in that language, if primary language is english description will be english and all programs are in the second language, other are ignore. 
So work around is if you have a language different of english you put it as primary and if you want only english you remove all secondary language and it works. 
[image]

mbonani, 2019-12-02:
can we do that from  https://bugreports.qt.io/browse/QTBUG-72491

> The solution has already been written.
> Just create * .proj under the "Resource" folder.
> (It seems that there is no need to put any files in the * .proj folder)
> 
> I have a project for MacAppStore with PySide2.
> I can not use macdeployqt.
> I've combined Py2app with my own script, but the workaround is the same, just create a folder.

mbonani, 2020-01-23:
fix in 286cbe775b33eba14e7aa8cc08989890ce51ac5e

closed 2020-01-23 by mbonani

mbonani, 2021-07-23:
this strange bug append one time on testing on a fresh installation on Mac OS. 
The computer was in English add second language in french. 
TS was in french and description in English
We switch to french as first language all was in french
We switch back in English all was in English
So there is a strange things how mac manage languages.

PR #635 (2019-09-05): Display the description in the preferred UI languages
Fixes #604