Mobsya/aseba #609: Text review in new Wireless configurator of Thymio Suite

cbarraud, 2019-08-28:
In the new configuration tools, I propose to do some text modifications:

### In the tools of Thymio Suite
[image]

- [ ] "Pair a Wireless Thymio to a Wireless dongle" 
- [ ] "Pair a group of Wireless Thymio robots independently"

### On the pop-up window before the configurator
[image]

- [ ] "Open the Wireless Thymio network configurator"
- [ ] "All Wireless Thymio robots associated to the Wireless dongle plugged on this computer will be disconnected until the Wireless Thymio network configuration window is closed."

### In the Pair a Thymio Wireless to a dongle interface
[image]

- [ ] "Wireless Thymio network configurator"
- [ ] "Plug the Wireless Thymio robot you want to pair using the USB cable"
- [ ] "Plug the Wireless dongle"
- [ ] "Please plug a Wireless Thymio robot with a USB cable to this computer"

### In the Advanced mode interface
[image]
- [ ] "Wireless Thymio network configurator"
- [ ] "To pair multiple Wireless Thymio robots to one Wireless dongle:"
- [ ] "03. Click on Pair !"

### In the pair a case of thymios interface
[image]

- [ ] "Wireless Thymio network configurator"
- [ ] "Plug the Wireless Thymio robot you want to pair using the USB cable"
- [ ] "Plug the Wireless dongle"
- [ ] "Tip: Put a sticker dot of the same color on both the Wireless Thymio robot and the Wireless dongle to identify them!
(open question, should we advise the color dot sticker or the serial number of the robot? could depends also of next move in production)

mbonani, 2019-08-28:
for me all comments are going to our standart. 
just for the case I am not sure if it is clear what means "Pair a group of Wireless Thymio robots independently""

mbonani, 2019-08-28:
>  (open question, should we advise the color dot sticker or the serial number of the robot? could depends also of next move in production)
to see with @riedo but basic idea was that we add stickers in the cases

cbarraud, 2019-08-28:
> just for the case I am not sure if it is clear what means "Pair a group of Wireless Thymio robots independently""

I admit that it's not very good :) a bit wordy too. 

"Pair several robots independently" ?
"Pair several Wireless Thymio robots each to one Wireless dongle" ?
"Pair a Wireless Thymio suitcase" ?

maybe the last one is good enough?

mbonani, 2019-08-28:
"Pair a Wireless Thymio case" we have to be simple
to be check but it seems "suitcase" is not the correct term in English, in french for sure it will be "Appairer une valise Wireless Thymio"

mbonani, 2019-08-28:
"Pair a case of Wireless Thymio" like in the PR is also good for me

PR #614 (2019-08-30): Improve texts
Fixes #609
commit 932bfa98 (2019-08-30): Improve texts

Fixes #609
closed 2019-08-30 by cor3ntin