Mobsya/aseba #620: Cannot Rename Blockly Variables

jmwiit, 2019-09-02:
When creating a variable in Blockly, we cannot get the dialog box to enter a new name for the variable when we click on the variable name. We get the drop-down menu, but when we select "rename variable" from the menu, no dialog box appears, so we cannot enter a new name. So far, we have been limited to a single variable named "item."

Are we missing something? Or is there a workaround?

mbonani, 2019-09-03:
Hi,
Can you give us your OS, if it is a mac can you give the version of OsX? On our computer this is working corretly.

jmwiit, 2019-09-03:
I am on Mac OSX 10.13.6.

Thanks,
  - Jim White

mbonani, 2019-09-03:
@cor3ntin can you control on your mac it's working on Sierra

jmwiit, 2019-09-03:
I just loaded up AsebaStudio, connected and turned on the Thymio robot, and started Blockly.

I created an event (just in case). 

Then I created a variable. Its default name was “item.” I clicked on the name “item,” and I got drop-down menu, with 3 choices: Item (the name of the variable), Rename variable…, and New variable….  The menu entry “item” was checked by default. I tried clicking on Rename variable…, but nothing happened. Nor did anything happen when I clicked on New variable….

I tried this on both the simple object for the variable, and also on the block for "Set item to …” Nothing happened in either case.

  - Jim

ypiguet-epfl, 2019-09-03:
Same problem in macOS 10.14.6 (Mojave). Same problem with the variable of the `for` loop. It must be the `prompt` function in `Blockly.FieldVariable.dropdownChange` which fails. Is there a way to display the debugger console in the browser of Thymio Suite?

cor3ntin, 2019-09-04:
To make that work on OSX and mobiles, we will need to replace te dialogs used by blockly -
Here is an example how to do that https://github.com/google/blockly/blob/7679d227df8f62a12316334ee708d4a9a800ef17/demos/custom-dialogs/custom-dialog.js

mbonani, 2019-09-13:
latest commit 09d3221 give a new version of blockly where you can change the name of variable. The "new variable" function do not work but the work around is to put a variable "item" from the menu and "rename"  it when needet. you can test on build  https://github.com/Mobsya/aseba/releases/tag/nightly

closed 2019-12-02 by mbonani