Mobsya/aseba #644: connecting with webapp is not working on macosx with remote robot

mbonani, 2019-09-09:
either chrome or safari, they cannot connect to robot on another computer in the network. 
I think it was working before 

[image]

perhaps parameters are not pass correctly?

cor3ntin, 2019-09-13:
Probably flatbuffers needing to be updated

mbonani, 2019-09-19:
this is consol from Chrome using Blockly, the Thymio is shown directly not connected

browser.js:14 file:///Applications/ThymioSuite.app/Contents/Resources/thymio_blockly/thymio_blockly.en.html
thymio_blockly.js:29 ws://localhost:8597 : undefined
thymio.js:516 connected, sending protocol version
thymio.js:536 Handshake complete: Protocol version 1

mbonani, 2019-09-19:
it seems that the TDM do not receive anything from Blockly:

[2019-09-19 08:33:08.819] [console] [debug]     aseba_endpoint.h@L207:  Message 'list nodes' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:08.820] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'node present' LÆopÚration a rÚussi
[2019-09-19 08:33:08.828] [console] [debug]     aseba_endpoint.h@L207:  Message 'get execution state' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:08.829] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'execution state' LÆopÚration a rÚussi
[2019-09-19 08:33:09.818] [console] [info]   aseba_endpoint.cpp@L261:   Requesting list nodes( ec : LÆopÚration a rÚussi )
[2019-09-19 08:33:09.819] [console] [debug]   aseba_endpoint.cpp@L274:  Waiting before requesting list node
[2019-09-19 08:33:09.820] [console] [debug]     aseba_endpoint.h@L207:  Message 'list nodes' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:09.820] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'node present' LÆopÚration a rÚussi
[2019-09-19 08:33:09.829] [console] [debug]     aseba_endpoint.h@L207:  Message 'get execution state' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:09.830] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'execution state' LÆopÚration a rÚussi
[2019-09-19 08:33:10.793] [console] [trace]       app_endpoint.h@L304:  <- Ping : LÆopÚration a rÚussi
[2019-09-19 08:33:10.819] [console] [info]   aseba_endpoint.cpp@L261:   Requesting list nodes( ec : LÆopÚration a rÚussi )
[2019-09-19 08:33:10.820] [console] [debug]   aseba_endpoint.cpp@L274:  Waiting before requesting list node
[2019-09-19 08:33:10.821] [console] [debug]     aseba_endpoint.h@L207:  Message 'list nodes' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:10.822] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'node present' LÆopÚration a rÚussi
[2019-09-19 08:33:10.829] [console] [debug]     aseba_endpoint.h@L207:  Message 'get execution state' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:10.830] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'execution state' LÆopÚration a rÚussi
[2019-09-19 08:33:11.820] [console] [info]   aseba_endpoint.cpp@L261:   Requesting list nodes( ec : LÆopÚration a rÚussi )
[2019-09-19 08:33:11.821] [console] [debug]   aseba_endpoint.cpp@L274:  Waiting before requesting list node
[2019-09-19 08:33:11.822] [console] [debug]     aseba_endpoint.h@L207:  Message 'list nodes' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:11.823] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'node present' LÆopÚration a rÚussi
[2019-09-19 08:33:11.830] [console] [debug]     aseba_endpoint.h@L207:  Message 'get execution state' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:11.830] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'execution state' LÆopÚration a rÚussi
[2019-09-19 08:33:12.822] [console] [info]   aseba_endpoint.cpp@L261:   Requesting list nodes( ec : LÆopÚration a rÚussi )
[2019-09-19 08:33:12.823] [console] [debug]   aseba_endpoint.cpp@L274:  Waiting before requesting list node
[2019-09-19 08:33:12.824] [console] [debug]     aseba_endpoint.h@L207:  Message 'list nodes' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:12.825] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'node present' LÆopÚration a rÚussi
[2019-09-19 08:33:12.831] [console] [debug]     aseba_endpoint.h@L207:  Message 'get execution state' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:12.832] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'execution state' LÆopÚration a rÚussi
[2019-09-19 08:33:13.295] [console] [trace]       app_endpoint.h@L304:  <- Ping : LÆopÚration a rÚussi
[2019-09-19 08:33:13.823] [console] [info]   aseba_endpoint.cpp@L261:   Requesting list nodes( ec : LÆopÚration a rÚussi )
[2019-09-19 08:33:13.824] [console] [debug]   aseba_endpoint.cpp@L274:  Waiting before requesting list node
[2019-09-19 08:33:13.825] [console] [debug]     aseba_endpoint.h@L207:  Message 'list nodes' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:13.826] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'node present' LÆopÚration a rÚussi
[2019-09-19 08:33:13.831] [console] [debug]     aseba_endpoint.h@L207:  Message 'get execution state' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:13.832] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'execution state' LÆopÚration a rÚussi
[2019-09-19 08:33:14.825] [console] [info]   aseba_endpoint.cpp@L261:   Requesting list nodes( ec : LÆopÚration a rÚussi )
[2019-09-19 08:33:14.825] [console] [debug]   aseba_endpoint.cpp@L274:  Waiting before requesting list node
[2019-09-19 08:33:14.826] [console] [debug]     aseba_endpoint.h@L207:  Message 'list nodes' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:14.827] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'node present' LÆopÚration a rÚussi
[2019-09-19 08:33:14.832] [console] [debug]     aseba_endpoint.h@L207:  Message 'get execution state' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:14.832] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'execution state' LÆopÚration a rÚussi
[2019-09-19 08:33:15.795] [console] [trace]       app_endpoint.h@L304:  <- Ping : LÆopÚration a rÚussi
[2019-09-19 08:33:15.826] [console] [info]   aseba_endpoint.cpp@L261:   Requesting list nodes( ec : LÆopÚration a rÚussi )
[... cut ...]
[2019-09-19 08:33:23.841] [console] [debug]     aseba_endpoint.h@L207:  Message 'list nodes' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:23.842] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'node present' LÆopÚration a rÚussi
[2019-09-19 08:33:23.842] [console] [debug]     aseba_endpoint.h@L207:  Message 'get execution state' sent : LÆopÚration a rÚussi
[2019-09-19 08:33:23.843] [console] [trace]   aseba_endpoint.cpp@L179:  Message received : 'execution state' LÆopÚration a rÚussi

mbonani, 2019-09-19:
this test are based on latest nightly build

mbonani, 2019-09-19:
same problem with build 66e9a01c50e7d2f40b6f0d9b55668cb3afb7b06d

mbonani, 2019-09-19:
also at that time aa2d1665d029fc6529900696832afd6fc4775163 webapp do not work with robot on another computer on Sierra with system browser instead of webview

mbonani, 2019-09-19:
I cannot go back more, I add no build

cor3ntin, 2019-09-23:
Fixed

closed 2019-09-23 by cor3ntin