Mobsya/aseba #653: while loop blocks events: unexpected behavior of Thymio - bug in the VM?

marcozeller, 2019-09-17:
In the following example Thymio never exits the while loop with a green top-LED:
```aseba
var a = 0

onevent button.center
	while a == 0 do
		call leds.top(0,255,0) # green
	end

onevent button.forward
	call leds.top(0,0,255) # blue
```
When pushing the forward button I would expect the top-LED to become blue.
I am using latest firmware and aseba 1.6.1 on Linux and a wireless Thymio (but the bug occurs as well when using the USB-cable)

Note that in the simulated environment (aseba-playground) this bug does NOT occur.

When making the loop condition interactive the behaviour becomes even weirder:
```aseba
onevent button.center
	call leds.top(255,0,0) # red
	while not button.backward  == 1 do
		call leds.top(0,255,0) # green
	end

onevent button.forward
	call leds.top(0,0,255) # blue
```
In this program the top-LED becomes green when pressing the center-button as expected.
But then when pressing the backward button it depends if the forward button has been pressed or not - either the top-LED becomes red or blue.
Both makes no sense to me...
Again in the simulated environment the program behaves just as I would have expected.

mbonani, 2019-09-17:
For me first you have to understand that the aseba VM is running on the robot or in the simulator, so the VM execution can be different even if normally it is the same VM, it is not the same processor, not exactly the same implementation because the hardware is different etc...
Also one thing is important is that the aseba VM execute event one after one, so they cannot run in parallel and if a VM is blocked by something that appens in another event then it block everything.
Aseba is made for executing thing quickly when one event occurs, and it should go out quickly of that event. In your example I cannot understand why you need the while loops? 
In fact it is what append in the simulator, the VM decide that this event has taken to much time and it is killed. 
[image]
So it is working in the simulator because the event is killed and the VM can treat the other events. Now why it is not working the same inside the robot I have to double check.  Normally this mechanism of killing event should also work...
More information about the aseba language and specially the events are here: http://wiki.thymio.org/en:asebalanguage#toc15

mbonani, 2019-09-20:
I confirm that blocking event are not killed in the real Thymio and this from very long time

PR #666 (2019-10-11): [VM] Kill events that take too many cycles to complete
Fixes #653
commit 80039948 (2019-10-11): [VM] Kill events that take too many cycles to complete

Fixes #653
closed 2019-10-11 by mbonani