Mobsya/aseba #703: on osx (10.14) animation and appliucation are no more working if you put on a second screen

mbonani, 2019-11-08:
sur mon macos 10.14.6 (18G103), su je déplace la fenêtre du launcher sur un deuxième écran, les animations se figent et on ne peut pas lancer les applications

cor3ntin, 2019-11-19:
I was not able to reproduce this issue (on 10.15)

mbonani, 2019-11-22:
I reproduce with 10.14.3

mbonani, 2019-11-22:
its work if you resize the windows, but when you drag from one screen to another one animation are freezed unti you rezie the windows.
When you have test have you rezise the window? Can we ack something around resizing, refereshing something?

mbonani, 2019-11-22:
or detecting with qt when you move the windows to refresh it also?
I did not find qt reletaed bug but probably I have not the correct spelling to found

cor3ntin, 2019-11-22:
no, I cannot fix this bug easily - i did not reproduce at all