Mobsya/aseba #704: scrolling on macs osx seems strange

mbonani, 2019-11-08:
le scroll de la fenêtre de choix des Thymio se comporte très bizarrement: pas de slider latéral tant qu’on ne va pas dessus, si on va dessus il n’apparait pas. Il apparait dès qu’on on scrolle avec le trackpad, mais avec la roulette du mouse le comportement est très bizarre: si tu vas en haut de la fenêtre et tu continues à scroller vers le haut, ensuite tu dois rescroller vers le bas un moment (sans que rien ne bouge) avant d’avoir les éléments qui scrollent. On dirait qu’il intègre le scroll au delà de la fin de page.

mbonani, 2019-11-08:
it seems ok on osx 10.12
should we let the scrollbar present if it is needet?

cor3ntin, 2019-11-15:
Should be better

closed 2019-11-18 by cor3ntin