Mobsya/aseba #707: arrow-buttons on the simulator on macOS

mbonani, 2019-11-12:
The arrow-buttons on the simulator on macOS are still not working (or just very occasionally).

mbonani, 2019-11-22:
it work on 10.12 
it is not working on 10.14 
opengl issue?

mbonani, 2019-12-10:
clicking on virtual button is working on 1.6.1 also on recent mac osx 10.14.

mbonani, 2019-12-10:
Here a small analysis of @stephanemagnenat :

> - Il y a du code picking [1]. La profondeur de la scène relative au point de click est prise dans le ZBuffer de OpenGL [2]. S'il y a intersection avec un objet de la scène, cet objet est retenu.
> 
> - Puis quand il y a un click [4], il est dispatché à cet objet.
> 
> - Le click arrive finalement dans le Thymio [5].

> Au vu de tes commentaires, j'imagine qu'il se situe dans l'accès au buffer OpenGL [2].

[1] https://github.com/enki-community/enki/blob/9e0f2efd608c796f0e756b6752fd8ba01b3d4578/viewer/Viewer.cpp#L997

[2] https://github.com/enki-community/enki/blob/9e0f2efd608c796f0e756b6752fd8ba01b3d4578/viewer/Viewer.cpp#L1022

[3] https://github.com/enki-community/enki/blob/9e0f2efd608c796f0e756b6752fd8ba01b3d4578/viewer/Viewer.cpp#L1035

[4] https://github.com/enki-community/enki/blob/9e0f2efd608c796f0e756b6752fd8ba01b3d4578/viewer/Viewer.cpp#L1223

[5] https://github.com/Mobsya/aseba/blob/4da4be90c67268919d1d3e2eccb9d2325f3adf5c/aseba/targets/playground/robots/thymio2/Thymio2.cpp#L78

mbonani, 2019-12-10:
Could also be related to this correction: 
https://github.com/Mobsya/enki/commit/0113f5b391adf37d2446efa759ebae2238b53460
or
https://github.com/Mobsya/enki/commit/6d0a122a2a5f3b53bc10b2e6585d18a86875bb1c

stephanemagnenat, 2019-12-12:
To add a comment to my private email to @mbonani, my guess is that on recent OS X releases that do not support OpenGL any more, Qt provides a polyfill, and that polyfill is buggy/incomplete, and does not allow to properly read back from the ZBuffer, which is what the picking code needs. The same polyfill might be the cause of crashes mentioned in the bugs above.

podak, 2020-01-23:
I tested it in macOS 10.15.1 and everything seems to be working correctly

mbonani, 2020-01-24:
user confirm that it is happening on 10.15 here is is configuration
[image]

mbonani, 2020-01-26:
problem with this mac,it will be update to 10.15 this week

[image]

KunjanChauhan, 2020-02-17:
Status agreed at 2020 02 17 meeting: 

- [ ] 1. so far unable to find a macine where problem can be duplicated.  Currently only seen in two users machines, both machines not accessible to dev teams.  Problems is seen on Catalina and MOjave versions of MACOS as well as Intel GPU and AMD Radeon GPUs, so likely a problem from OpenGL
- [ ] 2. Mobsya to try and find a machine that has same problem
- [ ] 3. Davide/Valentina to anlyse codebase

KunjanChauhan, 2020-02-18:
Fanny just tried the simulator on a MacBook Pro A1297 5,2 2009 with Nvidia GeForce 9600MGT  version running Mojave v10.14.2 and it works fine!

mbonani, 2020-02-18:
works on 
[image]

mbonani, 2020-02-18:
works on 
[image]

riedo, 2020-03-03:
I can reproduce the bug on my husband's computer. I could possibly use it to test when he doesn't need it - what is needed?
[image]

mbonani, 2020-03-03:
I have alos acces to my wife ones. I have install the compiling tools as explain in the building file.

mbonani, 2020-03-08:
works on 
[image]

mbonani, 2020-03-09:
**Last info** 
On one of the mac where it was not working, we compile on it and the simulator works normally. We try to reproduce that on the azure system and do not succeed with it. There are some problems: 
- passing from 10.13 to 10.14 was ok keeping xcode 10.1 (clang 10.0.1)
- Do not compile with xcode 10.3 (clang 10.1)
- Compile with xcode 11.2.1 (clang 11.0) but simulator do not work

actually on the special mac the build do not go trough every thing
- before installing xcode (clang 10.1) some error stop compilation for studio (same as azure)
- installing xcode 11.3.1 and 11.2.1 make compile more thing but linker blocks on TDM

mbonani, 2020-03-10:
finaly we found that the problem is because we declare the apps  defaults NSHighResolutionCapable in the .plist It seems for retina screen make the simulator not working well. Wthout declaring it OsX seems to manage better that. More test are needet to see the impat but proabably it will work because before this new packager we never make this add in .plist

mbonani, 2020-03-11:
fix in 43f25c2fac0ee1c95f2bbc0e2da939b2be4144b5

mbonani, 2021-11-03:
Some window user also have this issue if they have big resolution screen and zoom at 200%. Works when they put 100%.
It should be possible to calclutale better coordination.

mbonani, 2021-11-26:
probably we have to manage scale calculation like explin here https://vicrucann.github.io/tutorials/osg-qt-high-dpi/

mbonani, 2021-12-21:
fixed telling windows that Thymio Suite is DPI Unaware : **https://doc.qt.io/qt-5/highdpi.html**

closed 2021-12-21 by mbonani