Mobsya/aseba #712: flatpack-Thymio suite has not the correct icon (default one)

mbonani, 2019-11-15:
tested on Ubuntu 18.04

cor3ntin, 2019-11-15:
Do you have screens ? What is the correct icon, etc ?

mbonani, 2019-11-15:
for me it should have the thymio like on other OS

cor3ntin, 2019-11-15:
It is supposed to have this icon https://github.com/Mobsya/aseba/blob/master/menu/freedesktop/256x256/thymio-launcher.png

mbonani, 2019-11-15:
[image]

mbonani, 2019-11-15:
strange because when i make show details i have the correct icons:
[image]

KunjanChauhan, 2020-02-17:
Status agreed at 2020 02 17 meeting: 

- [ ] 1. Alessandra will place into PULL REQ by end Tuesday 18th
- [ ] 2. MB will test by end Wednesday 19th 
- [ ] 3. If testing is fine, update policy, finalize and merge into master - Alessandra?

mbonani, 2020-03-02:
flatpak version is 1.4.3

mbonani, 2020-03-02:
I update to 1.6.2 same behavior

ale-vit, 2020-03-02:
The version I tested is 1.6.1

mbonani, 2020-03-05:
fix in commit 43cfdcd

closed 2020-03-05 by mbonani