Mobsya/aseba #721: IOS scratch seems less reactive with infinite loop

mbonani, 2019-11-19:
I will put a test case

mbonani, 2019-11-22:
in fact it is less reactive also on desktop, you can bypass the problem by adding a small wait time in your loop like 0.1sec. In this case it is more reactive. Difficult to say if it is more visisble on tablet, probably a little. 

A small test program you can try is attached 
[test-reaction.zip](https://github.com/Mobsya/aseba/files/3879927/test-reaction.zip)

abbetch, 2019-11-27:
After de few tests  (ipad / thymio simulator) It seems that the longer the loop is running the less reactive it becomes. I presume that the motors start/stop  are stacking (since they don't seems to take time to be executed). Any other command that take times is fine. (like turn motor X for 1 sec).

abbetch, 2019-11-27:
I also tested with real devices. Using a cable it seems to work properly. With the wifi we can definitely see the issue (not as bad as in the simulator).
I can also confirm that the issue is also present using thymio suite on the desktop. So it does not seems related to IOS.

mbonani, 2019-11-27:
yes I agree related to https://github.com/Mobsya/scratch-vm/issues/15

closed 2021-12-13 by mbonani